koide3/glim #19: Posting the way your are using GLIM and related packages.

koide3, 2024-07-16:
TL; DR: Sharing that you are using GLIM (and related packages) helps the author receive recognition within his organization and enables continued work on this project.

Long description: The author of this package is NOT a full-time OSS developer but just a researcher who dedicates the majority of his working hours to other assignments. His time for OSS projects is limited, and to sustain his involvement, internal acknowledgment within his organization is crucial. Metrics like the number of stars and forks are often overlooked by executives, who may not grasp their significance. Conversely, stating "this package is used by A/B universities and C/D corporations" can effectively communicate its value. So, if you find this package useful for your work, we kindly ask you to leave a comment in this thread. Just saying hello to the author is appreciated. If you could share about your organization and how you use the package, that would immensely support the author.

Related: https://github.com/koide3/small_gicp/issues/3

robot-Yang, 2024-07-18:
Hello, koide-san, thank you for the wonderful work, I am interested in using GLIM for an indoor wheelchair navigation system, by the way, I am working at University of Tsukuba.

South-River, 2024-07-18:
hello，thank you for your great work.
i want to use it in multi robot collaborative mapping. by the way, the initialization and estimation of relative state might be a great idea in improving this work (like kimera-mmulti. as far as i know, there's not so much people working on this.
by the way, im working in zhejiang university

koide3, 2024-07-19:
@robot-Yang 
Thank you for your message! I'm so glad to hear from a good neighbor!

koide3, 2024-07-19:
@South-River 
Yeah, applications to multi robot settings must be an exciting topic. Actually, a student of mine is now working on multi-robot mapping, and I hope we can share interesting results with the community in the near future. Thanks for your comment!

changh95, 2024-07-22:
Thank you for another great contribution to the robotics community!

I'm trying to use GLIM and its related packages for my non-commercial personal project where I do LiDAR mapping of beautiful tourist places in my home country. I'll try it out with a number of LiDARs and mobile platforms, and in the future I will try to integrate into more robotics / autonomous driving projects.

I also plan to share the repo amongst my community of SLAM experts in Korea. There should be many who are working in the field of robotics and autonomous driving, both in academia and industry (I will definitely mention the commercial license).

koide3, 2024-07-22:
@changh95 
Thanks a lot! I hope this package will help your nice projects!

oliver-batchelor, 2024-07-28:
We (the University of Canterbury, New Zealand) attempt to map orchards with RTK GNSS, Mid360, and (12!) cameras. We use forms of photogrammetry (NeRF, Gassusian Splatting) and aim to align the sub-scans (e.g., from two sides of a row) using the LiDAR map (without it, there's a tendency to drift slightly).

koide3, 2024-07-29:
@oliver-batchelor 
Mapping with a dozen of cameras is interesting! Good luck on your work!

peter-ivarsen, 2024-08-26:
Hi, and thanks for sharing this project Koide-san. I work with a Norwegian reasearch institution and have tested GLIM in connection with a project we are part of,  on robotized ship recycling. So far I am very impressed by both the robustness and accuracy of this system.

fyandun, 2024-09-06:
Hi!

I'm from Carnegie Mellon University. I'm using your software for benchmarking different Lidar-inertial SLAM systems. Great work!

dezhi-l, 2024-09-30:
Thank you. awesome project !

hatem-darweesh, 2024-10-09:
Amazing work, I like the simplicity and easy to understand system of configuration files. 
I was using hdl_graph_slam for 3 years. currently I am moving to use GLIM. 
In my work I have used it for creating HD Maps for autonomous driving applications beside other confidential projects ;) 
I would love to see optimized parameters for indoor mapping.

se7oluti0n, 2024-10-24:
Hi!

Thank you for your great work. Among open-sourced 3D Lidar SLAM, I found that your work has the most well designed code base. The module odometry, submapping, global mapping and extension modules. All of that is very clever and make it easy to understand and extend.

Currently I use Glim for outdoor robot navigation. I have added relocalization, localization to GLIM. Also I added wheel odometry factor from @TakuOkawara san (https://github.com/TakuOkawara/full_linear_wheel_odometry_factor), and it solve the drift in the long narrow bridge 

Here is the video [https://youtu.be/qgoBgG9_V48](url)

koide3, 2024-10-29:
@se7oluti0n 
Thanks a lot for your comment and sharing the very nice video! I'll tell Taku that his package has been used in the community, and I'm sure he will be so happy!

oliver-batchelor, 2024-10-30:
> Hi!
> 
> Thank you for your great work. Among open-sourced 3D Lidar SLAM, I found that your work has the most well designed code base. The module odometry, submapping, global mapping and extension modules. All of that is very clever and make it easy to understand and extend.
> 
> Currently I use Glim for outdoor robot navigation. I have added relocalization, localization to GLIM. Also I added wheel odometry factor from @TakuOkawara -san (https://github.com/TakuOkawara/full_linear_wheel_odometry_factor), and it solve the drift in the long narrow bridge (see in the video below)
> 
> Here is the video [GLIM with localization and wheel odometry - Cong vien Thien Van Hoc](https://youtu.be/qgoBgG9_V48)

This is awesome - we might try to integrate this. We did some tests on a completely flat field to see how the GPS can be used effectively, and the odometry can fail. Though this is not our normal use case it seems like it should be useful to add for robustness anyway!

skohlbr, 2024-11-01:
Hi! Thanks for making GLIM available! We're currently evaluating it for use with inspection robots and things looks quite promising so far.

@se7oluti0n I was wondering if you plan on making your fork available, as your video looks great and we'd like to test those features, too.

se7oluti0n, 2024-11-02:
@skohlbr thanks for interested in my work. I hope I could publish some code soon. Which feature you are most interested in?

skohlbr, 2024-11-02:
@se7oluti0n Toss-up between relocalization and odometry integration (slight edge to relocalization as I believe the solution space might be bigger). Would be great to be able to check it out, but your call of course :)

oliver-batchelor, 2024-11-12:
> @skohlbr thanks for interested in my work. I hope I could publish some code soon. Which feature you are most interested in?

I'd be really keen to try wheel odometry - we'll be working on the same thing here at some point!

se7oluti0n, 2024-11-12:
@oliver-batchelor I have published the wheel odometry implementation. you can check glim, glim_ros2 repo in my page.

mamoto, 2024-11-24:
i used glim with LiDAR VLP-16, IMU VN-100s, Jetson Orin NX 16G configuration for very good data results, the software is awesome. i want to increase the number of images of the final map and add RGB data from the camera synchronized with the point cloud but i don't know how to configure? can you help me?

koide3, 2024-11-24:
@mamoto 
Thanks for your comment, and sorry, RGB data support is not included in this package. I think you can implement a some post process to generate a colored dense point cloud.

Efesendil, 2024-11-25:
> @mamoto Thanks for your comment, and sorry, RGB data support is not included in this package. I think you can implement a some post process to generate a colored dense point cloud.

Hello,
Thanks for this great work ! I've quickly tested the package with the given sensor data, I'd also like to know more about point cloud colorization tool since you created a chapter in the documentation about it. As you are saying it isn't included in this package means that there's a related package for colorization or is it under paid license ? If not implemented would you have any recommended packages to use in order to achieve that (Like lidar-camera calibration and writing pixels into scans) ? And are you thinking about to create such a package in future ?

mamoto, 2024-11-25:
> Hello, Thanks for this great work ! I've quickly tested the package with the given sensor data, I'd also like to know more about point cloud colorization tool since you created a chapter in the [...]

Thank you for your feedback on my question, I'm also looking into it to edit the source code you provided. I will try to finish it soon and send it to you. I think everyone can use it without having to pay. And this is the device I built.
[image]
[image]

Efesendil, 2024-11-25:
> Thank you for your feedback on my question, I'm also looking into it to edit the source code you provided. I will try to finish it soon and send it to you. I think everyone can use it without having [...]

Wow, seems great ! Definitly waiting for it ! I am really curious about the results, would be happy to keep in touch. I will see what i can do on my side, currently working on carla-sim.

koide3, 2024-11-26:
@Efesendil 
> As you are saying it isn't included in this package means that there's a related package for colorization or is it under paid license ? If not implemented would you have any recommended packages to [...]

The colorization module is maintained as a *closed* source code as described [here](https://koide3.github.io/glim/extensions.html#closed-source-extension-modules). I'm not planning to release such a package, but I hope someone will create and share it with the community. I designed GLIM to be easily extensible and released it with a permissive license, so I believe anyone can create and release such an extension module by themselves.

@mamoto 
I'm glad to see you trying to develop and share the module. It is what I want the community be like. Feel free to let me know when it gets advanced or  you face some technical troubles.

se7oluti0n, 2024-11-26:
@Efesendil 
Koide3-san also publish the lidar-calibration package here https://github.com/koide3/direct_visual_lidar_calibration . Hope you can use it and create a extension module colorization pointcloud.

eandert, 2024-12-03:
Thank you for releasing this package and all your hard work! This is one of the best SLAM packages I have used with LIDAR on a Jetson Orin. It doesn't consume all the CPU cores like many other SLAM packages and seems to run very comfortably on the Orin GPU. Have been using with a Hesai LIDAR and external IMU.

ArthurAlexandersson, 2024-12-10:
Thank you for providing such an amazing package!

yuguangxie, 2025-02-13:
Thank you for your open-source work. I am currently using it for my work.

WhiteCri, 2025-02-19:
Thank you for your great work. I'm working on agricultural robotics, and I am now trying to use GLIM to estimate the indoor drone odometry. I hope i could share my work gradly in the future with you~~

rimargo, 2025-03-11:
> Hi!
> 
> Thank you for your great work. Among open-sourced 3D Lidar SLAM, I found that your work has the most well designed code base. The module odometry, submapping, global mapping and extension modules. All of that is very clever and make it easy to understand and extend.
> 
> Currently I use Glim for outdoor robot navigation. I have added relocalization, localization to GLIM. Also I added wheel odometry factor from [@TakuOkawara](https://github.com/TakuOkawara) -san (https://github.com/TakuOkawara/full_linear_wheel_odometry_factor), and it solve the drift in the long narrow bridge (see in the video below)
> 
> Here is the video [GLIM with localization and wheel odometry - Cong vien Thien Van Hoc](https://youtu.be/qgoBgG9_V48)
> 
> **Update**: Integrate GPS Factor to align map to NED frame and help correct both horizontal and vertical drift. Video: https://www.youtube.com/watch?v=neGONn7brmI

Wow, localization to GLIM looks really good! What approach did you use to implement localization?

se7oluti0n, 2025-03-12:
@rimargo Thank you for your interest in my work. My approach is:
- Run Glim in mapping mode then use offline_viewer to manually adding the constraint to correct the map
- I mainly change the GlobalMapping module following these ideas:
- 1. Load the saved map but instead of restore the full factor graph, I create a GTSAM NonlinearEqualityFactor for each submap to freeze it when optimized.
- 2. The OdometryEstimation Module and Submapping module keep the same
- 3. For relocalization, I use the initilial pose, find the list of saved submap near the initial pose then select submaps have biggest overlap score to the current submap. Then find the exact pose by 3 level VGICP
- 4. After relocalization success, for each new submap, find the loop-closure constraint to saved submap by compare overlapscore the again VGICP.
- 5. Don't forget to reduce the PriorFactor confidence for first submap in GlobalMapping so FactorGraph able to corrrect pose in relocalization

raewon-sudo, 2025-04-01:
Thanks @koide3 for solution and GUI packages GLIM. 
@se7oluti0n Can you share source code for this: 

[image]

I want to reduce GPU memory usage during offline viewer processing.

marmik18, 2025-04-04:
@koide3 Thank you providing and maintaining such amazing packages.

@se7oluti0n even I am interested in the localization. Would it be possible for you to make the source code available?

koide3, 2025-04-04:
Hi guys,
I feel this thread has got a bit off topic. For a discussion about localization, please head to https://github.com/koide3/glim/issues/158

kamibukuro5656, 2025-04-17:
Thank you for your great work.
I always look forward to your work.
I have modified the moving point removal algorithm I created as a hobby to be able to load GLIM output.
It doesn't perform very well, but I'm happy if anyone is interested.

https://github.com/kamibukuro5656/MapCleaner_Unofficial

koide3, 2025-04-17:
@kamibukuro5656 
Thanks for your comment and for sharing your work! It looks so nice. I'll mention it in README!

hwan0806, 2025-07-27:
@koide3
The papers and codes you’ve shared have made a tremendous contribution to the LiDAR-based robotics perception community. I’ve been learning so much from your work ever since I was a student.

In my master’s project on multi-map alignment ([GitHub link](https://github.com/sparolab/uni-mapper)), I adopted the config utils and shared module functionalities from your ***glim*** project. These features are incredibly well-designed and have made development significantly more convenient.

You’ve been a real role model to me, and I’ll do my best to contribute meaningfully to this community as well. I’m very much looking forward to seeing more of your amazing work in the future.

Best regards,
Gilhwan

kadjor, 2026-08-17:
Hello Koide-san,

Thank you for GLIM — and for the care you put into it. It has made a real difference to a project you probably never imagined.

I develop Lidar Extractor, a free macOS and Windows app for the 3DMakerpro handheld scanners (Raven, Hawk, Eagle). My users are land surveyors, architects and contractors — most have never heard of SLAM or ROS. GLIM runs inside the app as an embedded binary, driven by a small feeder of about 165 lines that calls SubMapping and GlobalMapping directly, without ROS. They never see it. It just works.

On a real 22-minute, 652 m handheld indoor scan with a poorly closed loop, our previous in-house odometry took over 58 minutes. GLIM took 7 minutes and closed the loop to 6 cm. Indoors and in vegetation it is clearly the better engine, which is exactly where these scanners are used.

A user recently posted a scan of Dealey Plaza in Dallas to the 3DMakerpro group and said it was crisper than anything from the manufacturer's own software. That result is largely yours.

I credit you and your libraries here: https://lidar.kadjor.net/licences.php

Thank you again.
Aly
Lidar Extractor

koide3, 2026-08-17:
Hi @kadjor !
I'm very glad to hear that GLIM has been used in your project, and it's getting good feedback from your users!

ingmarsell, 2026-08-17:
Hi @koide3 , thanks for building GLIM.

We're using it as the reconstruction engine behind LidarFlow, a browser based tool that turns raw rosbag and MCAP recordings into georeferenced point cloud maps.

Users upload a recording, GLIM handles the SLAM reconstruction, and when GNSS is available, we use FlexCloud for georeferencing. The final output is PCD or PLY together with the trajectory.

GLIM is core infrastructure for us, so I wanted to say thanks directly instead of just quietly building on top of it.

Happy to share some reconstruction screenshots in Thirdparty results as well.

koide3, 2026-08-17:
Hi @ingmarsell ,
That's really great to hear, thanks for sharing! I'd love to see screenshots!

ingmarsell, 2026-08-18:
@koide3 , this is a reconstruction of a city drive from raw LiDAR data, processed with GLIM. Feel free to try it out at lidarflow.com. Any feedback is welcome!
[image]