koide3/glim #23: IMU Initialization

rafaelspring, 2024-07-16:
Hi Kenji, amazing project! Congratulations on the release!

I have a question: many popular visual SLAM methods (e.g. ORB-SLAM 3) have an IMU-initialization stage to get initial estimates for the initial gravity vector, initial velocities and IMU biases.
I haven't found anything similar in the GLIM paper or in the code.

Is there an initialization stage that I missed?
Or is GLIM making certain assumptions on the initial state and biases?

Kind regards,
Rafael

koide3, 2024-07-16:
There is [a loose coupling-based initial state estimation](https://github.com/koide3/glim/blob/master/src/glim/odometry/loose_initial_state_estimation.cpp) adopted from our following paper.

Tightly Coupled Range Inertial Localization on a 3D Prior Map Based on Sliding Window Factor Graph Optimization, ICRA2024

rafaelspring, 2024-07-16:
Thanks! I'll check it out.

closed 2024-07-17 by koide3

hunkyu, 2024-07-19:
@koide3 Tightly Coupled Range Inertial Localization on a 3D Prior Map Based on Sliding Window Factor Graph Optimization, ICRA2024 Will this part of the work be open source? Is it part of glim?

koide3, 2024-07-19:
The initialization algorithm is included in GLIM. The map-based localization algorithm will not be included.

rafaelspring, 2024-07-19:
Thanks, got it!

LitoNeo, 2024-08-18:
Great work! Thanks!