koide3/glim #30: link invalid

qiqzhang, 2024-07-20:
**Describe the bug**
wonderful job！I want use Offline viewer, but can't get data, can you resahre file, the link is invalid
dump_os1_128.tar.gz https://drive.google.com/file/d/1lOGGYpIicmBYZeaxKzwXc5NaK6d-krMn/view?usp=sharing

koide3, 2024-07-22:
Thanks for reporting the invalid link. I'll soon re-upload data and replace the link with a valid one.

koide3, 2024-07-23:
I uploaded an example dump data at: https://drive.google.com/file/d/1bZDhCTrhPftnU2d4UwRy3rhkbf2NI_eQ/view?usp=sharing

The doc will soon be updated. Note that I'm not 100% sure if the underlying boost serialization ensures the binary format stability and compatibility, and it may not be readable on environments with different boost versions (I used the one bundled with Ubuntu 24.04).

closed 2024-07-23 by qiqzhang