koide3/glim #31: warning: cudaErrorInsufficientDriver: CUDA driver version is insufficient for CUDA runtime version

gityeezy, 2024-07-21:
Hello, you did a great job. But when I was implementing the code, I encountered the problem of 'CUDA driver version is insufficient for CUDA runtime version'. My environment is using my ubuntu20.04 docker, using eigen3.4.0, cuda12.2.2, then following your tutorial to install gtsam, iridescence, gtsampoints, and finally using ros1 to encode the code. Although there were some problems in the middle, I solved them and the compilation passed. However, when I used the rosrun command, I started reporting errors. If I force run the bag package, it will be directly aborted. How should I find the problem?

[image]
[image]

South-River, 2024-07-21:
in my opinion, that's might because your gpu driver is too low that in sufficient for CUDAToolKit 12.2.2, maybe you should check your GPU driver and update it.

koide3, 2024-07-22:
As @South-River said, it might stem from your nvidia driver version. Check the driver version with `nvidia-smi`.

gityeezy, 2024-07-22:
Thank you, I have solved this problem. It is not because my driver version is too low or wrong, my driver is the latest, and my GPU supports cuda12.2. It is because my GPU computing power is 86, but during my compilation process, I set it to 89 by mistake, so the error is reported.

closed 2024-07-22 by gityeezy