koide3/glim #34: Slow processing speed

changh95, 2024-07-22:
**Describe the bug**

I ran GLIM on ROS1 Noetic docker image (koide3/glim_ros1:noetic_cuda12.2), with the provided downsized rosbag (os1_128_01_downsampled.bag).

It seems a lot slower than expected. I expected faster processing speed, from this video (https://www.youtube.com/watch?v=CIfRqeV0irE).

Is this processing speed normal?

Here's a video capture of my run.
https://youtu.be/q0Pk7DriU8s

**To Reproduce**
1. Pull docker image from `koide3/glim_ros1:noetic_cuda12.2`.
2. Start the docker container with `docker run -it --rm --net=host --ipc=host --pid=host --gpus all -e=DISPLAY -e=ROS_DOMAIN_ID -v ./config:/glim/config -v /data:/data koide3/glim_ros1:noetic_cuda12.2`
3. `roscore` + `rosrun glim_ros glim_rosnode` + `rosbag play /dta/os1_128_01_downsampled.bag`

**Expected behavior**
I expected faster processing speed, like https://www.youtube.com/watch?v=CIfRqeV0irE

**Screenshots**
https://youtu.be/q0Pk7DriU8s

**Environment (please complete the following information):**
 - CPU/GPU: Core i9-13900K, RTX3080
 - OS: Ubuntu 20.04
 - ROS: ROS1 noetic docker
 - CUDA: CUDA 12.2
 - Build with CUDA: ON

koide3, 2024-07-22:
To process faster, use `glim_rosbag` instead of `glim_rosnode`.

changh95, 2024-07-22:
> To process faster, use `glim_rosbag` instead of `glim_rosnode`.

Fixed!

`glim_rosbag` is super fast!! :+1: :+1: :+1:

changh95, 2024-07-22:
Closing issue as solved.

closed 2024-07-22 by changh95