koide3/glim #35: System block problem

South-River, 2024-07-22:
**Describe the bug**
I'm running your code based on Orbbec Femto Bold, which is a similar RGBD sensor to Azure Kinect.
But I found your code would report time gap between ros messages like 
```cpp
[2024-07-22 14:52:32.692] [warning] large time gap between consecutive IMU data!!
[2024-07-22 14:52:32.692] [warning] current=1721629076.512064 last=1721629076.329933 diff=0.182131
```
And the output of the slam would blocked when this message comes out.
The Environment I'm using is Ubuntu 20.04, ROS1 Noetic, GPU Nvidia 4060, CPU Intel i7-13700K, memory 64 GB. 
My camera sensor's cloud topic has 30 Hz, and IMU topic has 100 Hz.

One of a typical output would be like this, the system blocked and the wall mismatched (marked blue).
[image]

The output log show the imu data has large time stamp gap at time stamp 1721629148 (marked red in upper figure), but I checked the rosbag file and found both pointcloud data and imu data did not have large time stamp gap at that time stamp.
[image]
 

To solve the problem, I modified the source code in `glim_ros1/src/glim_rosnode.cpp`. My modification are listed below
```cpp
  GlimNode() : nh("~"), image_transport(nh) {
    ...
    // to avoid buffer too large
    image_sub = image_transport.subscribe(image_topic, 1, &GlimNode::image_callback, this);
    imu_sub = nh.subscribe(imu_topic, 10, &GlimNode::imu_callback, this, ros::TransportHints().tcpNoDelay());
    points_sub = nh.subscribe(points_topic, 1, &GlimNode::points_callback, this, ros::TransportHints().tcpNoDelay());
    ...
  }
...
  void points_callback(const sensor_msgs::PointCloud2ConstPtr& points_msg) {
    // typical Lidar message has 10 Hz, but my camera is 30 Hz
    static int32_t i=0; 
    i++;
    if (i%3 != 0)
      return;
     ...
  }
```
I also tried modify the config file `config_preprocess.json`
```
    "random_downsample_target": 5000,
    "random_downsample_rate": 0.3,
```
But the problem still occurs.
The rosbag and config I'm using could get from this link: http://zjufast.tpddns.cn:9110/share.cgi?ssid=62c30b2718fe450ab4153651c71e0397, meanwhile, I found the bag named 2024-07-22-14-19-50.bag would have more problems in both this issue and mapping results.

South-River, 2024-07-22:
Meanwhile, have you ever tested using azure kinect in outdoor scenario? I found it seems only can sense objects within 2~3m outdoor.

koide3, 2024-07-22:
Thank you for providing the bag files. I'll check them.

Yes, I've tested Azure Kinect in outdoor scenes, and it only provided poor point clouds because it relied on near-IR. I recommend using a LiDAR like Livox MID360 in outdoor situations.

koide3, 2024-07-23:
For `2024-07-22-14-17-05`, I didn't see major problems. It might be because I converted the bag into rosbag2. I'll test it on ROS1 later.

For `2024-07-22-14-19-50`, it seems there is a large IMU data drop that caused estimation corruption. I saw a similar problem when I used Azure Kinect at 30FPS that turned into a high bandwidth for bag recording. I recommend reducing the point cloud FPS and resolution (I obtained the best result with `fps=15` & `depth_mode=WFOV_2X2BINNED`) for both storage writing bandwidth and processing speed.

A small tip: For RGB-D cameras, set `sensors/global_shutter_lidar` to `true` to disable motion distortion correction.

South-River, 2024-07-23:
sincerely thank you for your help.
i also managed to run it in my environments with your tips ang get satifying results(๑ت๑). 
thanks for your tips and hoping for more interesting results in the fualso.∩_∩

closed 2024-07-23 by South-River

WhiteCri, 2024-12-20:
@South-River 
I'm also struggling with Femto + GLIM application...
Could you share your config files? My glim diverges with my configuration :(
Below is my configuration...
https://www.dropbox.com/scl/fo/ldvdkvkw7lemq73twbanw/AIT4JwpgvTvhdFCt6-v6c_E?rlkey=hp51ejpwosbddptcyrymb4cl8&st=9vd7s3sx&dl=0