koide3/glim #36: 'Open Map' in offline_viewer does nothing

changh95, 2024-07-23:
**Describe the bug**

The 'open map' button on the offline viewer does no action. (Happens in both ROS1 and ROS2)

Previously, it managed to open a window that allows me to explore directories.

I can only open the viewer by `ros2 run glim_ros offline_viewer /tmp/dump/`.

The application also does not have a help function - and I only found out this works by accident.

**To Reproduce**
Steps to reproduce the behavior:
1. Run GLIM
2. `ros2 run glim_ros offline_viewer` to open offline viewer.
3. Click on 'open map' on the left top side of the offline viewer application.
4. Nothing happens.

**Expected behavior**

A new window should open to explore directories.

**Screenshots**

https://youtu.be/HsYoDVVU5RA

**Environment (please complete the following information):**
 - CPU/GPU: Core i9-13900K, RTX 3080
 - OS: Host- Ubuntu 22.04, Docker image - koide3/ros2_humble_cuda12.2
 - ROS: ROS2 humble
 - CUDA: CUDA 12.2
 - Build with CUDA: ON

koide3, 2024-07-23:
It might stem from the underlying portable-file-dialog that assumes some window manager. Can you check if `sudo apt install zenity` fixes the issue?

koide3, 2024-07-23:
portable-file-dialog requires either `zenity` or `kdialog` to be installed to the system.
- GNOME desktop (using [Zenity](https://en.wikipedia.org/wiki/Zenity) or its clones Matedialog and Qarma)
- KDE desktop (using [KDialog](https://github.com/KDE/kdialog))

https://github.com/samhocevar/portable-file-dialogs

changh95, 2024-07-23:
[image]

It now works after `sudo apt install zenity` !

Also checked for Docker use cases. For docker, zenity install seems necessary.

changh95, 2024-07-23:
Closing the issue as now solved.

closed 2024-07-23 by changh95

koide3, 2024-07-23:
Yeah, it must be good to add `zenity` to dockerfiles. I'll do it later.