koide3/glim #37: Segmentation fault when using 'close map' without loading map on offline viewer

changh95, 2024-07-23:
**Describe the bug**

Not a critical error, but might be useful to report.

When using offline viewer, the user is expected to load a map, then later close the map.

If the user chooses to 'close map' without actually loading a map, the application closes with segmentation fault.

Probably a better way to handle is by showing an error message.

**To Reproduce**
Steps to reproduce the behavior:
1. Open offline viewer
2. Click on 'close map'
3. Application closes. On the terminal, there's segmentation fault.

**Expected behavior**

Application remains, probably with some error messages.

**Screenshots**

https://youtu.be/HsYoDVVU5RA

**Environment (please complete the following information):**
 - CPU/GPU: Core i9-13900K, RTX 3080
 - OS: Ubuntu 22.04, Docker koide3/ros2_humble_cuda12.2
 - ROS: ROS2 Humble
 - CUDA: CUDA 12.2
 - Build with CUDA: ON

koide3, 2024-07-23:
Thanks. Will fix it soon.

koide3, 2024-07-23:
https://github.com/koide3/glim/pull/39 should fix the issue and be merged into the main branch soon.

koide3, 2024-07-23:
Closing as the fix is merged.

closed 2024-07-23 by koide3