koide3/glim #44: Ready pointcloud with intensity

VanavaraDigital, 2024-07-25:
Hello, thank you for your great work.
I have one trouble:
Exported Pointcloud has no intensity value of points.
I learned source code and see, that frames points have intensity values, but next step, when you produce submaps from frames, submaps points have no intensity value.
Does it possible to add intensity values to points in exported pointcloud ?

Thanks

koide3, 2024-07-26:
The intensity information is lost in the frame merge in sub mapping. If you want intensity data, you need to modify the following part to retain the original intensity information.

https://github.com/koide3/glim/blob/3d65b3b72a730b6b41fc667ed639bae5c29ca78a/src/glim/mapping/sub_mapping.cpp#L461

F-lalala, 2024-08-07:
> The intensity information is lost in the frame merge in sub mapping. If you want intensity data, you need to modify the following part to retain the original intensity information.
> 
> https://github.com/koide3/glim/blob/3d65b3b72a730b6b41fc667ed639bae5c29ca78a/src/glim/mapping/sub_mapping.cpp#L461

Hello,I need to modify the function gtsam_points::merge_frames_auto to add intensity information,but I can't locate where the function was implemented, maybe I should rewrite it by myself?

koide3, 2024-08-08:
`merge_frames_*` are in [gtsam_points](https://github.com/koide3/gtsam_points/blob/f5eb8d3a7aa555d68801a66762136fc672cd3bca/src/gtsam_points/types/gaussian_voxelmap_cpu_funcs.cpp#L25). I'm thinking of implementing a merge function with intensity support, but it has a low priority for now.

F-lalala, 2024-08-09:
> `merge_frames_*` are in [gtsam_points](https://github.com/koide3/gtsam_points/blob/f5eb8d3a7aa555d68801a66762136fc672cd3bca/src/gtsam_points/types/gaussian_voxelmap_cpu_funcs.cpp#L25). I'm thinking of implementing a merge function with intensity support, but it has a low priority for now.

Thanks，I found the declaration of merge_frames_* in the file point_cloud_cpu.hpp，but it seems there is no corresponding function implemention in the file point_cloud_cpu.cpp.?

h-wata, 2024-11-21:
Thank you for your guidance on handling intensities with `gtsam_points`. Following your advice, I have implemented the following enhancements:

- **Intensity Handling in `gtsam_points`**: Modified `gtsam_points` to support intensity data.
- **Intensity Export in GLIM's Offline Viewer**: Updated GLIM's Offline Viewer to include intensity data when exporting maps, provided the submap contains intensity information.

These changes enable GLIM to process and export intensity data effectively. I've created branches in my forks for both repositories:

- GLIM: [https://github.com/h-wata/glim/tree/feature/export_ply_with_intensities](https://github.com/h-wata/glim/tree/feature/export_ply_with_intensities)
- gtsam_points: [https://github.com/h-wata/gtsam_points/tree/feature/add_intensities](https://github.com/h-wata/gtsam_points/tree/feature/add_intensities)

Feel free to explore these branches if you're interested.

3942368, 2025-09-08:
> Thank you for your guidance on handling intensities with `gtsam_points`. Following your advice, I have implemented the following enhancements:
>
> * **Intensity Handling in `gtsam_points`**: Modified [...]

I'm using the -DBUILD_WITH_CUDA=OFF option
Your modified gstam points fail to install because of an error in the make -j$(nproc) step, can you give me a solution?

https://github.com/koide3/gtsam_points
This installs and works fine, but the problem is that intensity doesn't work.