koide3/glim #53: Localization on Preloaded Map

Edwardius, 2024-07-29:
Hello, 

Is it possible to use GLIM to localize a robot on a preloaded map? My ideal is to not update the map during runtime.

Thanks

koide3, 2024-07-30:
No, localization on a prior map is not included in GLIM. We have another [closed code](https://koide3.github.io/glil_pubdoc/) for this purpose.

AphroraC, 2025-09-19:
> No, localization on a prior map is not included in GLIM. We have another [closed code](https://koide3.github.io/glil_pubdoc/) for this purpose.

Hello Dr.Koide! Thank you very much for making such an excellent work FOSS.
I am eager to gain a better understanding of your other project, GLIL, if possible.
I learned that GLIL is available under a paid license agreement.
If possible, could you please inform me with more information about the terms and conditions of the paid license?
Thank you again for your remarkable contributions to the field!