koide3/glim #57: glim_ros2 without viewer does not build

changh95, 2024-08-01:
In GLIM, there is an option to build without viewer (-DBUILD_WITH_VIEWER=OFF). This made me think there's a headless mode (i.e. no graphical user interface) for glim_ros2. 

When this option is disabled, glim library can be built, but the build for glim_ros2 fails. This means glim_ros2 forces the user to use the iridescence GUI.

By no means I'm not saying iridescence is bad (it's an amazing GUI!). But the conventional process of using glim_ros2 / glim_rosbag is to run the program, and then the user manually closes the iridescence window to get the pose and map output via dump. This makes it difficult to use for autonomous robot applications, where there should not be absolutely no manual user input.

I'm wondering if there is headless mode for glim_ros2, or I'm doing something wrong?

koide3, 2024-08-01:
Thank you for reporting the issue! Your feedback is always very clear and helpful to improve the package! I'll fix glim_ros2 to support building without viewer soon.  

Meanwhile, you can disable the iridescence viewer by removing `libstandard_viewer.so` in `config_ros.json`. I intentionally designed GLIM to be separable from the viewer to cover the use scenarios like you mentioned :)

koide3, 2024-08-01:
glim_ros2 now supports `-DBUILD_WITH_VIEWER=OFF` : https://github.com/koide3/glim_ros2/pull/27

changh95, 2024-08-02:
I tested this - it works nicely!

But without the viewer, it's a bit difficult to end the program to trigger dump process.
So it may be advised to auto-quit the program.

At the moment I changed the code for auto-quit to true, and it works flawlessly (https://github.com/koide3/glim_ros2/blob/973f1956d9090ae48d231aea1c01defce7fb1413/src/glim_rosbag.cpp#L195)

changh95, 2024-08-02:
Whether to make set auto-quit to true or not depending on the viewer build, is entirely design by choice in my honest opinion.
The issue is solved for me, so I'll close it at the moment.

If @koide3 thinks there could be more changes, feel free to re-open the issue :+1:

closed 2024-08-02 by changh95