koide3/glim #58: Add TF frames of our robot in config files

Gabryss, 2024-08-01:
**Is your feature request related to a problem? Please describe.**
I have a complete robotic stack working so far but I would like to change my SLAM algorithm for GLIM. However, it is not possible for me as GLIM create different TF frames as the ones I use (see the screenshot attached). I would then like to specify my own IMU frame and LiDAR frame and link the `odom` with my `base_footprint` frame.

**Describe the solution you'd like**
Would it be possible to specify a link to the Tf frame on the `config_ros.json` file or the `config_sensor.json` file. The ROS 2 convention ask for a link `map->odom->base_footprint` however some people use `map->odom->base_link`. Would it be possible to specify this in the config file ?

**Additional context**
In addition, I can not simply rename my current frames as this will break my other algorithm running on the rover.

[image]

koide3, 2024-08-02:
For now, to change TF frame names, you need to change the hardcoded names in https://github.com/koide3/glim_ros2/blob/973f1956d9090ae48d231aea1c01defce7fb1413/src/glim_ros/rviz_viewer.cpp#L18 . You can set a dummy frame name to `imu_frame_id` and connect it with `base_footprint`. 

I'm recognizing this a very poor way. It must be configurable. I'm thinking of adding the following parameters to the config to better support TF. Do you think it would suffice for your situation? 

- `imu_frame_id` : IMU frame or `auto` to automatically determine the name from IMU msg
- `lidar_frame_id` : LiDAR frame or `auto`
- `odom_frame_id` : Odom frame
- `world_frame_id` : Map frame
- `robot_root_frame_id` : Root frame of the robot (e.g., `base_footprint`). `root -> imu` transformation must be provided through TF so that GLIM can address the transformation.

Then, GLIM should publish a TF tree like `map -> odom -> robot_root` and `lidar -> imu`. I might be good to make `lidar -> imu` optional for the case the user already has a TF tree for this part.

koide3, 2024-08-02:
I added parameters to change TF frame names https://github.com/koide3/glim_ros2/pull/28 .

Gabryss, 2024-08-02:
Hello Koide,
Thank you so much, I can't wait to try this!