koide3/glim #63: How to use ros1 to run glim_ros1

hunkyu, 2024-08-14:
hunkyu@hunkyu:~/rs_ws/src/glim_ros1/launch$ roslaunch glim_rosnode.launch 
... logging to /home/hunkyu/.ros/log/71eb6650-5a28-11ef-8f71-cd42bf0a5762/roslaunch-hunkyu-45538.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hunkyu:45085/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    glim_rosnode (glim_ros/glim_rosnode)

ROS_MASTER_URI=http://localhost:11311

process[glim_rosnode-1]: started with pid [45552]
[ INFO] [1723636921.364012844]: Starting GLIM
[2024-08-14 20:02:01.365] [info] register linearization hooks
[2024-08-14 20:02:01.365] [info] config_path: /home/hunkyu/rs_ws/src/glim/config
[2024-08-14 20:02:01.365] [warning] param glim_ros/keep_raw_points not found
[2024-08-14 20:02:01.365] [warning] use default_value=false
[2024-08-14 20:02:01.366] [info] load libodometry_estimation_gpu.so
[2024-08-14 20:02:01.366] [warning] failed to open libodometry_estimation_gpu.so
[2024-08-14 20:02:01.366] [warning] libodometry_estimation_gpu.so: cannot open shared object file: No such file or directory
[2024-08-14 20:02:01.366] [critical] failed to load odometry estimation module
[glim_rosnode-1] process has died [pid 45552, exit code -6, cmd /home/hunkyu/rs_ws/devel/lib/glim_ros/glim_rosnode __name:=glim_rosnode __log:=/home/hunkyu/.ros/log/71eb6650-5a28-11ef-8f71-cd42bf0a5762/glim_rosnode-1.log].
log file: /home/hunkyu/.ros/log/71eb6650-5a28-11ef-8f71-cd42bf0a5762/glim_rosnode-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

hunkyu, 2024-08-14:
hunkyu@hunkyu:~/rs_ws$ rosrun glim_ros glim_rosnode
[ INFO] [1723637212.607151391]: Starting GLIM
[2024-08-14 20:06:52.607] [info] register linearization hooks
[2024-08-14 20:06:52.608] [info] config_path: /home/hunkyu/rs_ws/src/glim/config
[2024-08-14 20:06:52.608] [warning] param glim_ros/keep_raw_points not found
[2024-08-14 20:06:52.608] [warning] use default_value=false
[2024-08-14 20:06:52.608] [info] load libodometry_estimation_gpu.so
[2024-08-14 20:06:52.608] [warning] failed to open libodometry_estimation_gpu.so
[2024-08-14 20:06:52.608] [warning] libodometry_estimation_gpu.so: cannot open shared object file: No such file or directory
[2024-08-14 20:06:52.608] [critical] failed to load odometry estimation module

closed 2024-08-14 by hunkyu

MonsterQhz, 2025-02-23:
Dear man.Could please tell me how to solve this problem?