koide3/glim #64: Following the tutorial, the container could not be started

zhangzscn, 2024-08-15:
zhangzs@Precision-7960-Tower:~$ docker run   -it   --rm   --net=host   --ipc=host   --pid=host   --gpus all   -e=DISPLAY   -e=ROS_DOMAIN_ID   -v /mnt/b/代码/glim/config:/glim/config   koide3/glim_ros2:humble_cuda12.2   ros2 run glim_ros glim_rosnode --ros-args -p config_path:=/glim/config
[2024-08-15 07:40:19.070] [glim] [info] config_path: /glim/config
[2024-08-15 07:40:19.070] [glim] [info] load libodometry_estimation_gpu.so
warning: cudaErrorNoDevice
       : no CUDA-capable device is detected
warning: cudaErrorNoDevice
       : no CUDA-capable device is detected
warning: cudaErrorNoDevice
       : no CUDA-capable device is detected
[2024-08-15 07:40:19.074] [glim] [info] load libsub_mapping.so
warning: cudaErrorNoDevice
       : no CUDA-capable device is detected
[2024-08-15 07:40:19.075] [glim] [info] load libglobal_mapping.so
warning: cudaErrorNoDevice
       : no CUDA-capable device is detected
[2024-08-15 07:40:19.075] [glim] [info] load libmemory_monitor.so
[2024-08-15 07:40:19.075] [glim] [info] load libstandard_viewer.so
[2024-08-15 07:40:19.079] [glim] [info] load librviz_viewer.so
glfw error 65543: GLX: Failed to create context: BadValue (integer parameter out of range for operation)
[ros2run]: Segmentation fault

zhangzs@Precision-7960-Tower:~$ docker info
Client: Docker Engine - Community
 Version:    26.1.4
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.14.1
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.27.1
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 4
  Running: 0
  Paused: 0
  Stopped: 4
 Images: 6
 Server Version: 26.1.4
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local splunk syslog
 Swarm: inactive
 Runtimes: nvidia runc io.containerd.runc.v2
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: d2d58213f83a351ca8f528a95fbd145f5654e957
 runc version: v1.1.12-0-g51d5e94
 init version: de40ad0
 Security Options:
  apparmor
  seccomp
   Profile: builtin
  cgroupns
 Kernel Version: 6.5.0-45-generic
 Operating System: Ubuntu 22.04.4 LTS
 OSType: linux
 Architecture: x86_64
 CPUs: 112
 Total Memory: 250.7GiB
 Name: Precision-7960-Tower
 ID: 8441afb6-6b8d-40ef-adff-3734a93d7a4f
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Experimental: false
 Insecure Registries:
  registry.byd.com
  127.0.0.0/8
 Registry Mirrors:
  https://hub.uuuadc.top/
  https://docker.anyhub.us.kg/
  https://dockerhub.jobcher.com/
  https://dockerhub.icu/
  https://docker.ckyl.me/
  https://docker.awsl9527.cn/
  http://registry.byd.com/
 Live Restore Enabled: false

koide3, 2024-08-15:
Make sure that you installed [NVIDIA container toolkit](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html).

zhangzscn, 2024-08-16:
yes，I have installed

koide3, 2024-08-17:
Can you see nvidia driver information with the following command? If not, there should be some issues in docker setup.
`docker run --gpus all nvidia/cuda:12.2.0-devel-ubuntu22.04 nvidia-smi`

changh95, 2024-08-27:
Perhaps NVIDIA driver was not installed in the host system?
It seems like nvndia container toolkit is installed, according to `Runtimes: nvidia runc io.containerd.runc.v2`.