koide3/glim #65: Indeterminant linear system detected while working near variable

hardesh-peppermint, 2024-08-18:
Hello,

Firstly thank you for making this package.

**Describe the bug**
I'm trying to run the ROS2 package without IMU. Seeing the error in global optimization as described below.

**To Reproduce**
Set IMU to false and use the CPU method instead of GPU

**Expected behavior**
Error should not come

**Screenshots**
[image]

**Environment (please complete the following information):**
 - CPU/GPU: 12th Gen i7
 - OS: Ubuntu 22.04
 - ROS: ROS2 Humble
 - CUDA: None

koide3, 2024-08-21:
That error happens when IMU is disabled and there is absolutely no overlap between submaps. Please try the following configuration that force creating odometry-based relative pose constraints to make the system well constrained.

`config_global_mapping_gpu.json` : 
- `create_between_factors` : `true`
- `between_registration_type` : `NONE`

https://github.com/koide3/glim/blob/64a98d8ce59e83e07ae59e7765e656b677296c60/config/config_global_mapping_gpu.json#L33

changh95, 2024-08-27:
I'm getting a similar error, and I have IMU.
What could be the issue?

deliangye, 2024-09-12:
I run into the same issue with the data you provide -- os1_128_01. 
Reprodue: set IMU to true and use the CPU method instead of GPU

hardesh-peppermint, 2024-09-17:
Hi @koide3,
I'm using the CPU only method for now as I don't have GPU on the robot. 

Having overlap between submaps is understandable. Is it possible to integrate wheel odometry in the factor graph to handle this? I have really good wheel odometry and it's updated at 30 Hz

koide3, 2024-09-21:
Can anybody share a bag file with me to reproduce and fix the issue?

changh95, 2024-09-22:
> Can anybody share a bag file with me to reproduce and fix the issue?

I completely forgot to send!
I'll send you one tomorrow 😅😅😅

changh95, 2024-09-25:
@koide3 
Sent you the dataset to your email!
Bare in mind my dataset is quite long :sweat_smile: - there are 5 rosbags that needs to be played in order, using glim_rosbag. On the 4th rosbag, the reconstruction starts to break, despite good LiDAR overlaps and IMU sensor being present.
The error from GTSAM disappears when I used global mapping with `pose_graph` instead of just `gpu`.

koide3, 2024-10-08:
@changh95 
Thanks a lot for providing the rosbags! These were really helpful to find the cause of the issue.
I found that the issue stems from a long trajectory without loops that causes a numerical stability issue in global optimization.

I added a mechanism to fix the factor graph when an indeterminant linear system is detected.
I would be glad if you could check if the patch https://github.com/koide3/glim/pull/92 fixes the issue on your side.

I'm also trying to add a few mechanisms to further improve the stability of global mapping, and they will soon be added to https://github.com/koide3/glim/pull/92.

changh95, 2024-10-14:
@koide3 
Tested the code on my side (the same rosbag I sent you).
It works nicely!
The code can deal with the exceptions rising, and the factor graph management resets the state. With this, GLIM can go all the way to the end of the data sequence. :+1:

hardesh-peppermint, 2025-12-10:
Hi @koide3 ,

Thank you again for making this package. We briefly met at ICRA earlier this year.

I've resumed my work with this package again. The issue seems to be still there in the latest commit on main. I'm not using IMU and doing the optimization on CPU. Would be great if I can get help debug this issue. Let me know the best way to share some data to fix this issue.