koide3/glim #68: warning: cudaErrorIllegalAddress an illegal memory access was encountered

sczgzzzj, 2024-08-28:
Thank you for your work, using GPU can greatly improve the speed, but the bag running larger scenes will often prompt insufficient memory, I am 3060 12G graphics card, can you save the optimized results to the hard disk at the same time, to avoid the problem of insufficient memory, thank you

sczgzzzj, 2024-08-29:
According to your above settings, it really greatly reduces the memory consumption of GPU. Thank you very much. Can the offline_viewer of GLIM now add the function of Multiple map merging like interactive_slam? This function is very useful. It can merge several bags calculated separately.  Or can interactive_slam support GLIM output packets

sczgzzzj, 2024-08-29:
In addition, I would like to ask you if you can add 2 graphics cards and calculate and increase the memory at the same time for calculating large scenes.

koide3, 2024-09-03:
> Can the offline_viewer of GLIM now add the function of Multiple map merging like interactive_slam? 

It is in a future plan, but for now, no support on merging multi maps.

> I would like to ask you if you can add 2 graphics cards and calculate and increase the memory at the same time for calculating large scenes.

I expect drastic changes are required. Utilizing multi-GPUs for speeding up is relatively easy, but expanding the memory space is not. I think introducing some out-of-core GPU memory access mechanism should be a more realistic solution.