koide3/glim #73: run time error with cuda 12.6

zhangzscn, 2024-09-03:
ubuntu 22.04
cuda 12.6

ros2 run glim_ros glim_rosbag os1_128_01
/mnt/b/glim/install/glim_ros/lib/glim_ros/glim_rosbag: symbol lookup error: /usr/local/lib/libgtsam_points_cuda.so: undefined symbol: _ZNK5gtsam15NonlinearFactor5rekeyERKSt6vectorImSaImEE
[ros2run]: Process exited with failure 127

zhangzscn, 2024-09-03:
cuda 12.5 also the same error

koide3, 2024-09-03:
A similar error happens when GTSAM was rebuilt with different options. Can you try to clean install GTSAM and gtsam_points? I also recommend try installing PPA packages without building them by yourself.

https://koide3.github.io/glim/installation.html

zhangzscn, 2024-09-03:
sudo apt install -y libgtsam-points-cuda-12.5-dev
正在读取软件包列表... 完成
正在分析软件包的依赖关系树... 完成
正在读取状态信息... 完成                 
E: 无法定位软件包 libgtsam-points-cuda-12.5-dev
E: 无法按照 glob ‘libgtsam-points-cuda-12.5-dev’ 找到任何软件包

zhangzscn, 2024-09-03:
I also try to clean install GTSAM and gtsam_points,but it do not work

zhangzscn, 2024-09-03:
sorry, I git clone the gtsam,gtsam_points to the ros2_ws ,this make the error.just remove them form ros2_ws, solve the error

closed 2024-09-03 by zhangzscn