koide3/glim #76: colcon build "1 package failed: glim_ros"

alejandro-alonso-isi, 2024-09-17:
I have the same problem from of issue  and I checked a version of GTSAM how you sugere but is in  tag 4.2a9

I encountered the same issue "A build error when using colcon build in ROS2 #67" as described in a previous issue. I verified that the GTSAM tag is set to the version tag 4.2a9, but the problem persists. 
I have attached screenshots of the error for reference.

[image]
[image]

**Environment (please complete the following information):**
 - CPU/GPU: 13th Gen Intel(R) Core(TM) i7-13700/NVIDIA GeForce RTX 3060
 - OS: Ubuntu 22.04.4 LTS
 - ROS: Ros humble
 - CUDA: 12.2
 - Build with CUDA: OFF

koide3, 2024-09-21:
Can you check and try the following?
- CMake options for GTSAM were set as following [the installation instruction](https://koide3.github.io/glim/installation.html)
- GTSAM and gtsam_points were placed outside ros2_ws and built and installed to the system
- Check if demo programs in gtsam_points work
- Run `sudo ldconfig` after installing GTSAM and gtsam_points

UriyMihailov, 2024-09-26:
I have the same problem. I've tried calling make for gtsam_points with build demos option, but all of them failed with a lot of undefined references. But if I call make without this option - gtsam_points builds just fine.

UriyMihailov, 2024-09-26:
Is it possible it has something to do with the CUDA option?

UriyMihailov, 2024-09-26:
[image]
[image]
[image]
[image]

alejandro-alonso-isi, 2024-10-01:
Now when I try to install gtsam_points, I get the following error mentioning an incompatibility with the Eigen version. I followed the step-by-step instructions as provided in the README
[image]

koide3, 2024-10-13:
These errors are typically caused by installing a wrong version of GTSAM. Please check the installed GTSAM version with:

```bash
ls /usr/local/lib/libgtsam*
# /usr/local/lib/libgtsam_points_cuda.so  /usr/local/lib/libgtsam.so    /usr/local/lib/libgtsam.so.4.2.0     /usr/local/lib/libgtsam_unstable.so.4
# /usr/local/lib/libgtsam_points.so       /usr/local/lib/libgtsam.so.4  /usr/local/lib/libgtsam_unstable.so  /usr/local/lib/libgtsam_unstable.so.4.2.0
```

If there exist `libgtsam.so.4.3a0` or something similar, remove them and re-install **GTSAM 4.2a9**.

ikhann, 2025-02-05:
@UriyMihailov @alejandro-alonso-isi Hi guys!
Have you resolved this issue? I'm experiencing the same problem. GTSAM 4.2a9.

gfr-misha, 2025-07-02:
For anyone still struggling, for me, the issue comes from ROS having its own GTSAM, so when you source ROS before colcon build, the wrong GTSAM version is used. I simply did `sudo apt remove ros-jazzy-gtsam`  and it built no problem.