koide3/glim #84: Localization only mode?

lisosia, 2024-09-25:
Is it possible to run glim in a localization only mode as follows:
1. After creating the map, save the map and exit.
2. Load the map and perform localization only (no map update).
I plan to use glim with MID-360.
Sorry if this is a strange question.

Thanks.

koide3, 2024-09-26:
We have another package GLIL for prior-map-based localization, but it's closed source requiring a proprietary license.

lisosia, 2024-09-30:
I got it. Thank you for your answer!

closed 2024-09-30 by lisosia

zhoubo2018, 2025-05-16:
> We have another package GLIL for prior-map-based localization, but it's closed source requiring a proprietary license.

Hello,how to get this closed source? Can you give me some introduction about that(GLIL) localization system?