koide3/glim #86: Clarification on LiDAR tilt configuration in GLIM

wsung1, 2024-10-05:
Hello, koide3-san,

First, I want to express my appreciation for your incredible work!

I’m currently using GLIM to generate HD maps for my mobile robot, which is equipped with a LiDAR (OS1-64) and a built-in IMU. To expand the field of view onto the ground, I have tilted the LiDAR along the y-axis (approximately 30 degrees nose-down pitch).

I believe that the LiDAR pose should be reflected in the transformation T_world_lidar. From what I understand, this is automatically configured by T_world_imu = T_world_lidar * T_lidar_imu, as mentioned in the config_sensors.json file. Since T_lidar_imu is provided and T_world_imu can be estimated using the gyroscope, am I correct in this understanding?

If I’m mistaken, could you please advise on what steps I should take when using a tilted LiDAR with GLIM?

Thank you!

koide3, 2024-10-07:
Yes, it's correct. `T_world_lidar` is a transformation that brings a LiDAR point to the world frame, and is derived as `T_world_lidar = T_world_imu * T_lidar_imu.inverse()`. The initial sensor pose is automatically determined such that the gravity direction (upward direction) of the IMU is aligned with the Z-direction of the world frame, so you don't need a special care for a tilted LiDAR.

wsung1, 2024-11-12:
I apologize for the delayed response, koide3-san.
I understand T_world_lidar = T_world_imu * T_lidar_imu.inverse but my observations do not align with this understanding.
Although my LiDAR is tilted nearly 30 degrees nose-down pitch, I didn't need to revise 'config_sensors.json' because, as we discussed, T_world_lidar should be automatically calculated to account for the LiDAR tilt and compensate for it.
However, in practice, the raw point cloud is tilted as much as the LiDAR itself (see Figure 1).
[image]
If the automatic calculation were correct, the point cloud should be parallel to the ground (indicated by the grey line in the middle).
The resulting map also appears tilted (see Figures 2 and 3).
[image]
[image]
Given that T_world_lidar is not correct, does this indicate a problem with the measurement of T_world_imu?
Is there a way I can debug this, possibly through the log entries shown in Figure 1:
[odom][info] T_world_imu
[odom][info] imu_bias
For a shift, the point cloud could be parallel to the ground by manipulating LiDAR config in 'config_sensors.json'
from
"T_lidar_imu" : [
0.006,
-0.012,
0.008,
0.0,
0.0,
0.0,
1.0
],
to
"T_lidar_imu" : [
0.006,
-0.012,
0.008,
0.0,
0.43,
0.3,
0.9
],
but this is nothing but a makeshift.
Your insights would be greatly appreciated.
Thank you.

skohlbr, 2024-11-17:
You still need to provide a correct transform for `T_lidar_imu` indicating the transform between LIDAR and IMU in the config. If that is set correctly, everything should be working as @koide3 described.