koide3/glim #87: /opt/ros/humble/lib/glim_ros/glim_rosnode: symbol lookup error: /usr/local/lib/libgtsam_points.so: undefined symbol: _ZNK5gtsam15NonlinearFactor5rekeyERKSt6vectorImSaImEE

Shaswat2001, 2024-10-07:
I installed GLIM using the PPA method for ROS2 humble on Ubuntu 22.04, but when I run `ros2 run glim_ros glim_rosnode`,

I get the error - 

/opt/ros/humble/lib/glim_ros/glim_rosnode: symbol lookup error: /usr/local/lib/libgtsam_points.so: undefined symbol: _ZNK5gtsam15NonlinearFactor5rekeyERKSt6vectorImSaImEE

How to solve this??

koide3, 2024-10-07:
Did you run `sudo ldconfig` after installing the package?

valgur, 2024-10-07:
I would suggest running the executable with LD_DEBUG=all environment var set and sharing the output here or in a gist. That might hopefully provide sufficient detail to pinpoint the specific issue.

thisismoiself, 2024-10-08:
I get the same error on Ubuntu 20.04 using ROS1. 
Executing `rosrun glim_ros glim_rosnode` works fine, but after launching the rosbag with `rosbag play --clock os1_128_01.bag`, the same error occurs and the node crashes:

`/opt/ros/noetic/lib/glim_ros/glim_rosnode: symbol lookup error: /opt/ros/noetic/lib/libglim.so: undefined symbol: _ZNK12gtsam_points6KdTree10knn_searchEPKdmPmPd`
`/opt/ros/noetic/lib/glim_ros/glim_rosnode: symbol lookup error: /opt/ros/noetic/lib/libglim.so: undefined symbol: _ZNK12gtsam_points6KdTree10knn_searchEPKdmPmPd
`

Shaswat2001, 2024-10-09:
I was able to solve this issue by doing - 

`export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH`

closed 2024-10-09 by Shaswat2001

thisismoiself, 2024-10-22:
Unfortunately, my error still persists: 
`/opt/ros/noetic/lib/glim_ros/glim_rosnode: symbol lookup error: /opt/ros/noetic/lib/libglim.so: undefined symbol: _ZNK12gtsam_points6KdTree10knn_searchEPKdmPmPd`

I am running Ubuntu 20.04 with ROS1, CUDA 12.5 and a clean install from the PPA. I would be grateful for any help.