koide3/glim #88: Direclty save pcd file

Shaswat2001, 2024-10-07:
Hi, 

Is it possible to directly save the pointcloud map created as PCD file rather than exporting it via the OfflineViewer?

koide3, 2024-10-07:
It is not directly supported, but you can easily implement it by using `global_mapping->export_points()` at the [destructor of `GlimROS`](https://github.com/koide3/glim_ros2/blob/15f05ec4a4d70da9152fe2e1bd3eeafb28001b93/src/glim_ros/glim_ros.cpp#L187).

```cpp
std::vector<Eigen::Vector4d> points = global_mapping->export_points();
```

Shaswat2001, 2024-10-07:
Thanks. And sorry, this question might be stupid. But can I make this change if the glim_ros library is installed using PPA method?

Shaswat2001, 2024-10-09:
Thanks a lot for the help. I was able to make the change.

closed 2024-10-09 by Shaswat2001

hatem-darweesh, 2024-10-11:
@Shaswat2001  You can save as .ply using offline_viewer. 
Then convert and visualize it to (.pcd) format using python open3d. 

```
import open3d as o3d 
pcd = o3d.io.read_point_cloud("source_pointcloud.ply")
o3d.io.write_point_cloud("sink_pointcloud.pcd", pcd)
o3d.visualization.draw_geometries([pcd])
```

nazneenkhan4, 2025-03-13:
I tried using this method 
 std::vector<Eigen::Vector4d> points = async_global_mapping->export_points();

When I called it using a ros service it fails and returns no points but when I call it in the end in the destructor it works. Can someone tell me why?