koide3/glim #89: gtsam_points/include/gtsam_points/cuda/kernels/vgicp_derivatives.cuh(123): error: calling a __host__ function("Eigen::MatrixBase<.

hiroki-hashimoto-24, 2024-10-07:
**Describe the bug**
I tried to install GLIM by following the install from source in this URL (https://koide3.github.io/glim/installation.html). However, an error occurred when I tried to make gtsam_points. How can I resolve this error?

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/koide3/gtsam_points
2. mkdir gtsam_points/build && cd gtsam_points/build
3. cmake .. -DBUILD_WITH_CUDA=ON
4. make -j$(nproc)

**Expected behavior**
/home/hashimoto/gtsam_points/include/gtsam_points/cuda/kernels/vgicp_derivatives.cuh(123): error: calling a __host__ function("Eigen::MatrixBase< ::Eigen::CwiseBinaryOp< ::Eigen::internal::scalar_sum_op<float, float> , const  ::Eigen::Matrix<float, (int)3, (int)3, (int)0, (int)3, (int)3> , const  ::Eigen::Matrix<float, (int)3, (int)3, (int)0, (int)3, (int)3> > > ::inverse() const") from a __device__ function("gtsam_points::vgicp_error_kernel::operator () const") is not allowed

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Environment (please complete the following information):**
 - CPU/GPU: [Intel(R) Core(TM) i7-7700K/NVIDIA GeForce GTX 1080 Ti]
 - OS: [Ubuntu 22.04]
 - ROS: [ROS2 humble]
 - CUDA: [CUDA 12.2]
 - Build with CUDA: [ON]

**Additional context**
I am using Google Translate to create this article.

koide3, 2024-10-07:
It seems the build system is using a manually installed Eigen (possibly older than Eigen 3.4.0) at `/usr/local/include` (It should be located at `/usr/include` usually). 

Please try

1. Check the version of system installed Eigen:
```bash
$ grep -r VERSION /usr/include/eigen3/Eigen/src/Core/util/Macros.h 
#define EIGEN_WORLD_VERSION 3
#define EIGEN_MAJOR_VERSION 4
#define EIGEN_MINOR_VERSION 0
```

2. Check the version of manually installed Eigen:

```bash
$ grep -r VERSION /usr/local/include/eigen3/Eigen/src/Core/util/Macros.h 
```

3. Remove the manually installed Eigen if its version is older than 3.4.0:

hiroki-hashimoto-24, 2024-10-07:
Thank you for your reply. 
My Eigen version was 3.4.0. What should I check next in this case?
[image]