koide3/glim #90: Reducing weight of the IntegratedVGICPFactorGPU factor?

zMkOrg, 2024-10-07:
Hi, thank you for sharing this great work. 
I want to reduce the weight of the IntegratedVGICPFactorGPU factor so I can integrate data from other high precision sensors. Could you tell me how to do this? I would be very grateful. I didn't see any function interfaces in gtsam_points for reducing weight or increasing noise.

koide3, 2024-10-07:
While gtsam_points doesn't have a direct interface to do it, you can change the matching weight by increasing the covariance of each point (`Frame::covs`).

zMkOrg, 2024-10-12:
it worked! thank you for your kind help and immediate response.

> While gtsam_points doesn't have a direct interface to do it, you can change the matching weight by increasing the covariance of each point (`Frame::covs`).

closed 2024-10-15 by zMkOrg