koide3/glim #91: Build error in ROS2

Shaswat2001, 2024-10-07:
I tried building glim_ros2 on Ubuntu 22.04 and ROS2 humble. But get the following error 

Cmake Error

Starting >>> glim_ros
--- stderr: glim_ros                             
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::JacobianFactor::updateHessian(std::vector<unsigned long, std::allocator<unsigned long> > const&, gtsam::SymmetricBlockMatrix*) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::FixedLagSmoother::eraseKeyTimestampMap(std::vector<unsigned long, std::allocator<unsigned long> > const&)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::NonlinearFactor::rekey(std::vector<unsigned long, std::allocator<unsigned long> > const&) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::EliminateableFactorGraph<gtsam::GaussianFactorGraph>::marginalMultifrontalBayesNet(boost::variant<gtsam::Ordering const&, std::vector<unsigned long, std::allocator<unsigned long> > const&>, std::function<std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::GaussianFactor> > (gtsam::GaussianFactorGraph const&, gtsam::Ordering const&)> const&, boost::optional<gtsam::VariableIndex const&>) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::HessianFactor::HessianFactor(std::vector<unsigned long, std::allocator<unsigned long> > const&, std::vector<Eigen::Matrix<double, -1, -1, 0, -1, -1>, std::allocator<Eigen::Matrix<double, -1, -1, 0, -1, -1> > > const&, std::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > const&, double)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::PrintKeyVector(std::vector<unsigned long, std::allocator<unsigned long> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (unsigned long)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/offline_viewer.dir/build.make:272: offline_viewer] エラー 1
gmake[1]: *** [CMakeFiles/Makefile2:281: CMakeFiles/offline_viewer.dir/all] エラー 2
gmake[1]: *** 未完了のジョブを待っています....
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::FixedLagSmoother::eraseKeyTimestampMap(std::vector<unsigned long, std::allocator<unsigned long> > const&)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::HessianFactor::HessianFactor(std::vector<unsigned long, std::allocator<unsigned long> > const&, std::vector<Eigen::Matrix<double, -1, -1, 0, -1, -1>, std::allocator<Eigen::Matrix<double, -1, -1, 0, -1, -1> > > const&, std::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > const&, double)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::PrintKeyVector(std::vector<unsigned long, std::allocator<unsigned long> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (unsigned long)> const&)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::JacobianFactor::updateHessian(std::vector<unsigned long, std::allocator<unsigned long> > const&, gtsam::SymmetricBlockMatrix*) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::EliminateableFactorGraph<gtsam::GaussianFactorGraph>::marginalMultifrontalBayesNet(boost::variant<gtsam::Ordering const&, std::vector<unsigned long, std::allocator<unsigned long> > const&>, std::function<std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::GaussianFactor> > (gtsam::GaussianFactorGraph const&, gtsam::Ordering const&)> const&, boost::optional<gtsam::VariableIndex const&>) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::NonlinearFactor::rekey(std::vector<unsigned long, std::allocator<unsigned long> > const&) const'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/glim_rosnode.dir/build.make:271: glim_rosnode] エラー 1
gmake[1]: *** [CMakeFiles/Makefile2:200: CMakeFiles/glim_rosnode.dir/all] エラー 2
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::JacobianFactor::updateHessian(std::vector<unsigned long, std::allocator<unsigned long> > const&, gtsam::SymmetricBlockMatrix*) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::FixedLagSmoother::eraseKeyTimestampMap(std::vector<unsigned long, std::allocator<unsigned long> > const&)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::NonlinearFactor::rekey(std::vector<unsigned long, std::allocator<unsigned long> > const&) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::EliminateableFactorGraph<gtsam::GaussianFactorGraph>::marginalMultifrontalBayesNet(boost::variant<gtsam::Ordering const&, std::vector<unsigned long, std::allocator<unsigned long> > const&>, std::function<std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::GaussianFactor> > (gtsam::GaussianFactorGraph const&, gtsam::Ordering const&)> const&, boost::optional<gtsam::VariableIndex const&>) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::HessianFactor::HessianFactor(std::vector<unsigned long, std::allocator<unsigned long> > const&, std::vector<Eigen::Matrix<double, -1, -1, 0, -1, -1>, std::allocator<Eigen::Matrix<double, -1, -1, 0, -1, -1> > > const&, std::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > const&, double)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::PrintKeyVector(std::vector<unsigned long, std::allocator<unsigned long> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (unsigned long)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/validator_node.dir/build.make:271: validator_node] エラー 1
gmake[1]: *** [CMakeFiles/Makefile2:254: CMakeFiles/validator_node.dir/all] エラー 2
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::FixedLagSmoother::eraseKeyTimestampMap(std::vector<unsigned long, std::allocator<unsigned long> > const&)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::HessianFactor::HessianFactor(std::vector<unsigned long, std::allocator<unsigned long> > const&, std::vector<Eigen::Matrix<double, -1, -1, 0, -1, -1>, std::allocator<Eigen::Matrix<double, -1, -1, 0, -1, -1> > > const&, std::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > const&, double)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::PrintKeyVector(std::vector<unsigned long, std::allocator<unsigned long> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (unsigned long)> const&)'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::JacobianFactor::updateHessian(std::vector<unsigned long, std::allocator<unsigned long> > const&, gtsam::SymmetricBlockMatrix*) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::EliminateableFactorGraph<gtsam::GaussianFactorGraph>::marginalMultifrontalBayesNet(boost::variant<gtsam::Ordering const&, std::vector<unsigned long, std::allocator<unsigned long> > const&>, std::function<std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::GaussianFactor> > (gtsam::GaussianFactorGraph const&, gtsam::Ordering const&)> const&, boost::optional<gtsam::VariableIndex const&>) const'
/usr/bin/ld: /usr/local/lib/libgtsam_points.so: undefined reference to `gtsam::NonlinearFactor::rekey(std::vector<unsigned long, std::allocator<unsigned long> > const&) const'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/glim_rosbag.dir/build.make:271: glim_rosbag] エラー 1
gmake[1]: *** [CMakeFiles/Makefile2:227: CMakeFiles/glim_rosbag.dir/all] エラー 2
gmake: *** [Makefile:146: all] エラー 2
---
Failed   <<< glim_ros [2.44s, exited with code 2]

Summary: 0 packages finished [2.94s]
  1 package failed: glim_ros
  1 package had stderr output: glim_ros

I have followed the exact steps in the installation page. But the issue still exists. How to solve it?

Shaswat2001, 2024-10-09:
I guess this was something related to my system only. Because I was able to build the package on a different machines successfully.

koide3, 2024-10-13:
These errors are typically caused by installing a wrong version of GTSAM. Please check the installed GTSAM version with:

```bash
ls /usr/local/lib/libgtsam*
# /usr/local/lib/libgtsam_points_cuda.so  /usr/local/lib/libgtsam.so    /usr/local/lib/libgtsam.so.4.2.0     /usr/local/lib/libgtsam_unstable.so.4
# /usr/local/lib/libgtsam_points.so       /usr/local/lib/libgtsam.so.4  /usr/local/lib/libgtsam_unstable.so  /usr/local/lib/libgtsam_unstable.so.4.2.0
```

If there exist `libgtsam.so.4.3a0` or something similar, remove them and re-install **GTSAM 4.2a9**.

ooww0123TW, 2025-04-09:
In my case, this problem was related to libmetis-dev.

When compiling gtsam, although default set was using 3rdparty metis in gtsam repository, system metis was linked when generating libgtsam.so (you can check this using `$ ldd libgtsam.so`)

After I remove libmetis (`$ sudo apt remove --purge libmetis-dev`) to prevent gtsam using system metis, and re-compile gtsam / gtsam_points, I could solve the problem.

The strange thing is at first I didn't have this problem... I don't get it why this bug happens..

hoangsep, 2025-04-11:
@ooww0123TW I believe you can also use the system metis by changing the flag in `cmake/HandleMetis.cmake`
I had a similar issue with Eigen

HappySamuel, 2025-05-07:
> These errors are typically caused by installing a wrong version of GTSAM. Please check the installed GTSAM version with:
> 
> ls /usr/local/lib/libgtsam*
> # /usr/local/lib/libgtsam_points_cuda.so  /usr/local/lib/libgtsam.so    /usr/local/lib/libgtsam.so.4.2.0     /usr/local/lib/libgtsam_unstable.so.4
> # /usr/local/lib/libgtsam_points.so       /usr/local/lib/libgtsam.so.4  /usr/local/lib/libgtsam_unstable.so  /usr/local/lib/libgtsam_unstable.so.4.2.0
> If there exist `libgtsam.so.4.3a0` or something similar, remove them and re-install **GTSAM 4.2a9**.

I tried to check with `$ ls /usr/local/lib/libgtsam*` , and everything is like this:
```
/usr/local/lib/libgtsam_points.so  /usr/local/lib/libgtsam_unstable.so
/usr/local/lib/libgtsam.so         /usr/local/lib/libgtsam_unstable.so.4
/usr/local/lib/libgtsam.so.4       /usr/local/lib/libgtsam_unstable.so.4.2.0
/usr/local/lib/libgtsam.so.4.2.0
```

However my issue still remain
`symbol lookup error: /lib/aarch64-linux-gnu/libgtsam_unstable.so.4: undefined symbol: _ZNK5gtsam14JacobianFactor13updateHessianERKSt6vectorImN3tbb13tbb_allocatorImEEEPNS_20SymmetricBlockMatrixE`