koide3/glim #94: /opt/ros/noetic/lib/glim_ros/glim_rosnode: symbol lookup error: /opt/ros/noetic/lib/libglim.so: undefined symbol: _ZNK12gtsam_points6KdTree10knn_searchEPKdmPmPd

thisismoiself, 2024-10-22:
When following the "Getting Started" page on the wiki for ROS1, after running `roscore`, `rosparam set use_sim_time true`, `rosrun glim_ros glim_rosnode`, when playing the rosbag with `rosbag play --clock os1_128_01_downsampled.bag`, the node crashes with this error: 
`/opt/ros/noetic/lib/glim_ros/glim_rosnode: symbol lookup error: /opt/ros/noetic/lib/libglim.so: undefined symbol: _ZNK12gtsam_points6KdTree10knn_searchEPKdmPmPd`

[image]

I am also using `export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH` from issue #87 

I am running Ubuntu 20.04, ROS Noetic (ROS1), CUDA 12.5 and a clean PPA install of glim for ROS1 and CUDA 12.5.

Any help would be greatly appreciated!

sponsoredlinks, 2024-10-25:
I'm having the same issue with the non-cuda variant.

[unverified] My favourite AI suggests that it might be a mismatch between default params in the interface:

In cloud_preprocessor.cpp, the function is being called without the max_sq_dist parameter:

cppCopytree.knn_search(points[i].data(), k, k_indices.data(), k_sq_dists.data());

Change it to include the parameter:
cppCopytree.knn_search(points[i].data(), k, k_indices.data(), k_sq_dists.data(), std::numeric_limits<double>::max());

And add the necessary include at the top:
cppCopy#include <limits>

This should resolve the undefined symbol error because now it will match the exact function signature that exists in the library:
Copy00000000001a1fb0 T gtsam_points::KdTree::knn_search(double const*, unsigned long, unsigned long*, double*, double) const

The mismatch happened because while the header declares a default parameter, at the binary/linking level C++ requires an exact match of the function signature including all parameters.
[/unverified]

koide3, 2024-10-29:
The error stems from the inconsistency of library versions. glim_ros1 in the PPA uses gtsam_points v1.0.3 that has a slight API change from the latest v1.0.5. I'll soon update glim_ros1 PPA so that it uses the latest gtsam_points.

For now, you can try installing gtsam_points=1.0.2 to avoid the error.

koide3, 2024-10-29:
I just updated the PPA. Please try installing the latest `glim_ros1=1.0.5`.

sponsoredlinks, 2024-10-29:
Thanks @koide3 ! Do we do this via apt? It appears to be stuck on 1.03 for noetic:

user@host:~$ sudo apt update
Hit:1 http://au.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://au.archive.ubuntu.com/ubuntu focal-updates InRelease
Hit:3 http://au.archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:4 http://security.ubuntu.com/ubuntu focal-security InRelease
Hit:5 http://packages.ros.org/ros/ubuntu focal InRelease
Hit:6 http://download.opensuse.org/repositories/devel:/kubic:/libcontainers:/stable/xUbuntu_20.04  InRelease
Hit:7 https://apt.kitware.com/ubuntu focal InRelease
Ign:8 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/7.0 InRelease
Hit:9 https://koide3.github.io/ppa/ubuntu2004 ./ InRelease
Hit:10 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/7.0 Release
Reading package lists... Done
Building dependency tree
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
user@host:~$ sudo apt install --upgrade -y ros-noetic-glim-ros
Reading package lists... Done
Building dependency tree
Reading state information... Done
ros-noetic-glim-ros is already the newest version (1.0.3-0focal).
The following package was automatically installed and is no longer required:
  librhash0
Use 'sudo apt autoremove' to remove it.
0 to upgrade, 0 to newly install, 0 to remove and 1 not to upgrade.

koide3, 2024-10-29:
Sorry, I forgot to commit changes. I just re-pushed updated packages.

sponsoredlinks, 2024-10-29:
Thanks - do you know am I doing something wrong still here?

rosrun glim_ros glim_rosbag /temp/fast-lio-2024-10-26-09-58-00.bag 
[ INFO] [1730242549.835713843]: Starting GLIM
[2024-10-30 09:55:49.837] [info] register linearization hooks
[2024-10-30 09:55:49.844] [info] config_path: /opt/ros/noetic/share/glim/config
[2024-10-30 09:55:49.845] [info] load libodometry_estimation_ct.so
[2024-10-30 09:55:49.846] [info] load libsub_mapping.so
[2024-10-30 09:55:49.847] [warning] param sub_mapping/submap_target_num_points not found
[2024-10-30 09:55:49.847] [warning] use default_value=-1
[2024-10-30 09:55:49.847] [info] load libglobal_mapping.so
[2024-10-30 09:55:49.847] [warning] Extension modules are enabled!!
[2024-10-30 09:55:49.847] [warning] You must carefully check and follow the licenses of ext modules
[rospack] Error: package 'glim_ext' not found
[librospack]: error while executing command
[2024-10-30 09:55:49.860] [info] config_ext_path: /config
[2024-10-30 09:55:49.860] [info] load libmemory_monitor.so
[2024-10-30 09:55:49.861] [info] load libstandard_viewer.so
[2024-10-30 09:55:49.870] [info] load librviz_viewer.so
/opt/ros/noetic/lib/glim_ros/glim_rosbag: symbol lookup error: /opt/ros/noetic/lib/librviz_viewer.so: undefined symbol: _ZN4glim17TrajectoryManagerC1Ev

koide3, 2024-10-30:
Possibly, `ros-noetic-glim-*` still remains `1.0.3`. Can you try upgrading it to `1.0.5`?

thisismoiself, 2024-10-31:
> Sorry, I forgot to commit changes. I just re-pushed updated packages.

That solved at least my issue, thank you so much! Keep up the great work! 
Should i close this issue or leave it open for now?

koide3, 2024-10-31:
Good. Please close the issue.

closed 2024-10-31 by thisismoiself