koide3/glim #96: The package can not build with error CUDA::cudart

hoangkhai6893, 2024-10-29:
This is my error  show when I try to  build the package : 
" CMake Error at /usr/local/lib/cmake/gtsam_points/gtsam_points-targets.cmake:69 (set_target_properties):
  The link interface of target "gtsam_points::gtsam_points_cuda" contains:

    CUDA::cudart

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/local/lib/cmake/gtsam_points/gtsam_points-config.cmake:53 (include)
  CMakeLists.txt:30 (find_package)"
Currently, I using Jetson Nano to build this package. And setting on CmakeList.txt file is using build with CUDA.
Could  you help me to fix it?

koide3, 2024-10-29:
What is the version of your ubuntu and ROS? If it is older than ubuntu 22.04, you need to upgrade the cmake version.

hoangkhai6893, 2024-10-29:
I am using Ubuntu 22.04 for Jetson. The  OS for Jetson  has name Linux ubuntu 5.15.148-tegra with docker name is nvcr.io/nvidia/l4t-base:35.3.1. 
Cmake version am using is 3.24

koide3, 2024-10-30:
I'm not sure but it seems like there are some issues on cmake and CUDA. Did you see any other warning/errors with cmake?

hoangkhai6893, 2024-10-31:
Thank you for your support. Finally, I resolved the problem. I added `find_package(CUDAToolkit REQUIRED)` and `find_package(CUDA REQUIRED)` to CMakeLists.txt.
The problem comes from the Docker Container that I use on Jetson.