koide3/glim #102: Reproduce Flatwall experiment

se7oluti0n, 2024-10-30:
Hello, first of all, thank you for generously opensource this repository.

From your paper and the project page, I understand that Glim is able to recovery from lidar degeneration. I try to verify that using the flatwall_01 data but couldn't able to reproduce the recovery. Below is my configuration, could you show me which is wrong in my configuration. 
[config_sub_mapping_cpu.json](https://github.com/user-attachments/files/17568250/config_sub_mapping_cpu.json)
[config_global_mapping_cpu.json](https://github.com/user-attachments/files/17568252/config_global_mapping_cpu.json)
[config_odometry_cpu.json](https://github.com/user-attachments/files/17568253/config_odometry_cpu.json)
[config_ros.json](https://github.com/user-attachments/files/17568254/config_ros.json)
[config_sensors.json](https://github.com/user-attachments/files/17568255/config_sensors.json)

koide3, 2024-10-31:
The odometry estimation module is the main part to deal with point cloud degeneration, and for the flatwall experiment, you need to use OdometryEstimationGPU. You can find the configuration file [here](https://staff.aist.go.jp/k.koide/projects/glimsupp/flatwall.html). Note that the framework has been largely updated since the first release, and it may show slightly different results for a few sequences.

se7oluti0n, 2024-10-31:
Thanks for the prompt reply. Close threat here.

closed 2024-10-31 by se7oluti0n