koide3/glim #104: glfw error 65539

hoangkhai6893, 2024-10-31:
Hi everyone, I run the glim_rosnode then I got the errors print in my terminal
`glfw error 65539: Invalid window attribute 0x0002000D` . 
[image]

How can I resolve this problem? Thank you so much

koide3, 2024-10-31:
Are you running it natively or on docker? If the latter, you may need some X11-related docker options. If the former, please check if the GPU is visible to the system with `nvidia-smi`.

hoangkhai6893, 2024-10-31:
I am running it natively on Jetson Orin Nano with Ubuntu 22.04 version and CUDA 12.2. This is my result of nvidia-smi: 
[image]. 
Do you think the problem comes from glfw lib?