koide3/glim #108: Loop closure doesn't fix the Z-axis drift. Any ideas?

se7oluti0n, 2024-11-07:
Hello,

First of all, thank you for the contribution.
I have tested mapping on 1 round of the Park near my house. The park size is about 500m x 280m. 
I the the elevation of the Park is almost the same, some point on the trajectory has higher elevation, few meter.

For the map, there is one loop closure because I went 1 round of the park. I used offline viewer to create manual factor.
But the final map after the loop-closure show huge different of elevation (about 30 meter) at two side of the park?

Any ideas to improve the z-drift after loop closure?
Below is the picture of the map after loop - closure.
**Update**: I added GPS-factor to submapping and global mapping. With the good gps like in the park, there is almost no XY drift. But the Z-drift remaining the same.
[image]
[image]

tahakenar, 2024-11-07:
Hi, 
I am also facing the same issue with a relatively small map (~20m x ~170m road). I minimized cumulative Z-drift by manually closing the loop. Yet, there is still difference in Z axis between two ends of the map.

I provide the screenshots below. With no loop closure, Z-drift after closure and final map after loop closure respectively.

[image]
[image]
[image]

se7oluti0n, 2024-11-07:
@tahakenar Thank for the information. One thing that is similar in our cases is we both traveled in a quite narrow road with the high tree at the sides at some points. May that case cause the error in the pitch angle estimation and lead to the z drift, I guess.

tahakenar, 2024-11-08:
@se7oluti0n I processed my bag file to filter out point cloud data.  I clamped points using boundaries -15m to 15m for x and y. In z axis I clamped between -1.5m to 2.5m. Seems like it yields better results in terms of z-axis drift.

Before finding overlapping submaps and manual loop closure:

[image]

After manual loop closure and finding overlapping submaps:

[image]

[image]

skohlbr, 2024-11-08:
Depends on the use case, but the https://github.com/koide3/glim_ext?tab=readme-ov-file#flat-earther plugin might solve this issue if the environment is approximately flat.

koide3, 2024-11-10:
Yes, the [flat_earther plugin](https://github.com/koide3/glim_ext?tab=readme-ov-file#flat-earther) should solve the issue. In particular, if your environment is approximately flat overall, setting `max_neighbor_distance` in `config_flat_earther.json` to a large value (e.g., 1000 m) forces the map to be flat.

se7oluti0n, 2024-11-11:
@skohlbr @koide3 Thanks for the suggestion. Closed topic here.

closed 2024-11-11 by se7oluti0n