koide3/glim #110: Outdoor driving test with Livox MID360

zwqily, 2024-11-07:
Hello, may I ask if there is any demonstration data for the "Outdoor driving test with Livox MID360" in your documentation? I am learning about the processing of vehicular LiDAR and would appreciate your assistance. Thank you.

koide3, 2024-11-10:
I'll upload a MID360-demo data next week. Meanwhile, please see [the setup guide](https://github.com/koide3/glim/wiki/Sensor-setup-guide) for the configuration for MID360.

ikhann, 2025-02-08:
Hi @koide3 

Thank you a lot for this remarkable project!

I wanted to kindly follow up on this request. Is there any available demonstration data for the outdoor test with Livox MID360?

Thank you!

koide3, 2025-02-10:
Sorry, I completely forgot this. I just uploaded the example dataset. Please see here: https://koide3.github.io/glim/demo.html#outdoor-driving-test-with-livox-mid360

ikhann, 2025-02-10:
> Sorry, I completely forgot this. I just uploaded the example dataset. Please see here: https://koide3.github.io/glim/demo.html#outdoor-driving-test-with-livox-mid360

So fast! Thank you very much!