koide3/glim #112: fatal error: gtsam_points/config.hpp: No such file or directory

JeremyLebon, 2024-11-07:
**Describe the bug**

When building glim and glim_ros2 with ./glim_ros2/docker/deb/Dockerfile.ros2 dockerfile. 

I now get a fatal error gtsam_points/config.hpp: No such file or directory. In the past I could just build the glim (v1.0.4)  and glim_ros2. I updated to 1.0.6 but still the same error

I can see some changes in the commit of the repo gtsam_points. Could they be causing the problem? How do I link when building to an older version of gtsam_points.

[image]

I don't know if this linked but when running `ros2 run glim_ros glim_rosnode` in the `koide3/glim_ros2:jazzy` directly for the docker hub I get the following.
[image]

**Environment (please complete the following information):**
 - CPU/GPU: Intel(R) Xeon(R) W-10885M CPU @ 2.40GHz/Quadro RTX 4000
 - OS: Ubuntu 24.04 in docker container
 - ROS: ROS 2 Jazzy
 - CUDA: 12.5

koide3, 2024-11-10:
`gtsam_points>=1.0.5` introduced `config.hpp` to manage build options, and `glim>=1.0.5` relies on this feature. Please try upgrading `gtsam_points` to the latest one, or downgrading `glim` to 1.0.4.

`koide3/glim_ros2:jazzy` is built without CUDA support. Use `koide3/glim_ros2:humble_cuda12.2` instead.

JeremyLebon, 2025-02-04:
Last week I rebuild my docker container and everything seems to building and beter more working !! :)  Very nice stuff!

closed 2025-02-04 by JeremyLebon