koide3/glim #116: imu_sursor is not used in loose_initial_state_estimation.cpp

KOKIAOKI, 2024-11-21:
**Describe the bug**
Hello! koide3! Thank you for your support always.

I wonder why imu_cursor is not used for imu_data in this [line](https://github.com/koide3/glim/blob/master/src/glim/odometry/loose_initial_state_estimation.cpp#L133).

```
const Eigen::Vector3d acc_local = imu_data[i].middleRows<3>(1);
```

**To Reproduce**
Is the following correct?
```
const Eigen::Vector3d acc_local = imu_data[imu_cursor].middleRows<3>(1);
```

Then, run to check
``` 
ros2 run glim_ros glim_rosbag [bag file]
```

**Expected behavior**
I think this will correct the estimation of the direction of gravitational acceleration while moving.

**Screenshots**
I haven't noticed any changes in the test data, but I'll update if I find any changes in other data.

**Environment (please complete the following information):**
 - CPU/GPU: Corei7-13700H (laptop)
 - OS: ubuntu22
 - ROS: ROS2 humble
 - CUDA: CUDA12.2
 - Build with CUDA: ON

**Additional context**

koide3, 2024-11-24:
Thanks for reporting the issue. Yes, it should use `imu_cursor` instead of `i`. I'll fix it soon. Fortunately, this bug should only have a small impact because `sum_acc_odom` is used for only rough initial estimation of the gravity direction.

koide3, 2024-11-26:
I just merged a PR to fix this issue https://github.com/koide3/glim/pull/119. Thanks again for your report.