koide3/glim #124: IMU-Lidar timestamp issue

Efesendil, 2024-12-05:
Hello,
I am getting inconsistent (the map drifts after a while) results while using imu and lidar together in carla sim, what could be the issue for the warnings below ? 

[2024-12-05 01:03:42.608] [odom] [warning] insufficient number of IMU data between LiDAR scans!! (odometry_estimation)
[2024-12-05 01:03:42.608] [odom] [warning] t_last=179.920230 t_current=179.970230 num_imu=1
[2024-12-05 01:03:42.681] [odom] [warning] insufficient number of IMU data between LiDAR scans!! (odometry_estimation)
[2024-12-05 01:03:42.681] [odom] [warning] t_last=179.970230 t_current=180.020230 num_imu=1
[2024-12-05 01:03:42.751] [odom] [warning] insufficient number of IMU data between LiDAR scans!! (odometry_estimation)
[2024-12-05 01:03:42.751] [odom] [warning] t_last=180.020230 t_current=180.070230 num_imu=1

ooww0123TW, 2024-12-17:
I also always get this error at the end of the bag file..

Efesendil, 2024-12-17:
The problem is in sub_mapping.cpp from line 217 to 242, where the imu preintegration was handled. Probably it is waiting for more frequent imu data in the defined interval (50 miliseconds in my case since the lidar frequency is at 20 hz). However i don't know how to configure carla simulator to publish more frequent imu data than lidar data. 
To be able to successfully handle the imu preintegration we should have at least 2 imu inputs in this interval, as we can see from the warnings i only have one, i will try to modify the code to set this number to 1 in order to see if the imu data will be added as a factor into the graph and if the drift still occurs (since carla is a simulation environment, even with 1 imu data it should handle correctly, my imu noise rates are null).  
However i don't have a clear solution for real hardware if this problem occurs, probably it is something to do with sensors hardware synchronization and time stamping. Most of the time IMU publishes data at 100hz and lidar at 20hz, so it should work :)

ooww0123TW, 2025-04-07:
How frequently your imu and lidar data published?