koide3/glim #125: use offline_viewer to registrate multi-view point clouds

CharlieV5, 2024-12-05:
Hi, Koide-san!
I am impressed with the effect of global optimization with multiple loop constraints. And I want to test if the multi-view TLS( Terrestrial Laser Scanner ) point clouds  can be registated well by GLIM offline_viewer.
So I write some code to make the dump data for offline_viewer as follow.
[test_global_mapping.zip](https://github.com/user-attachments/files/18016899/test_global_mapping.zip)
[test-data.zip](https://github.com/user-attachments/files/18016914/test-data.zip)
The test-data contains 8 stations of TLS point cloud, and the pose of each station. The test_global_mapping will convert the test-data into dump data for offline_viewer. In the code, I convert the xyz clouds into bin files and construct the factor graph for the multiple stations.
Then I get the dump data, the offline_viewer can load the data and show the right multi-view point clouds. But after choosing optimization and finding overlapping submap, it crashed.

Any advice to make it possible to registrate multi-view cloud using GLIM?

https://github.com/user-attachments/assets/9c6559a2-e155-4bd5-a706-06529d69b323

koide3, 2024-12-29:
Wow, it's an interesting trial. I guess we need an additional prior factor to fix the gauge freedom of the entire graph. If you could share the dump file, I would like to try implementing a mechanism to detect and fix the exception.

CharlieV5, 2024-12-30:
> Wow, it's an interesting trial. I guess we need an additional prior factor to fix the gauge freedom of the entire graph. If you could share the dump file, I would like to try implementing a mechanism to detect and fix the exception.

OK，this is the dump file for you.
https://drive.google.com/file/d/1GeabUtvAkszC1CVOchTDegXcK1r-NoAx/view?usp=drive_link

koide3, 2025-01-08:
I found that the error is caused by submap point cloud data that contains corrupted covariance matrices. I just pushed a patch to detect and fix corrupted point cloud https://github.com/koide3/glim/pull/131.

CharlieV5, 2025-01-10:
Thank you, Koide-san! I've tested the latest code with the dump data. It works great! I will test with more TLS data later.

koide3, 2025-01-10:
Your welcome! BTW, feeding TLS data to GLIM would be of interest to people. Do you mind if I share the mapping result of your dump data on SNS (X)?

CharlieV5, 2025-01-11:
> Your welcome! BTW, feeding TLS data to GLIM would be of interest to people. Do you mind if I share the mapping result of your dump data on SNS (X)?

Not at all. Feel free to share the data!

closed 2025-01-26 by CharlieV5