koide3/glim #127: It shows [ros2run]: Segmentation fault when running in ROS2 in local environment

shavargo109, 2024-12-11:
**Describe the bug**
Following the 'Installation' and 'Getting started' guide, everything installed OK. When running "ros2 run glim_ros glim_rosnode", it shows[ros2run]: Segmentation fault, and cannot proceed afterward

**To Reproduce**
Steps to reproduce the behavior:
1. ros2 run glim_ros glim_rosnode in the workspace

**Expected behavior**
works normally followed the guide

**Screenshots**
There is some problem on uploading screenshot, I will post the log in terminal instead

ros2 run glim_ros glim_rosnode 
[2024-12-11 17:51:58.173] [glim] [info] config_path: /home/asd/glim_ws/install/glim/share/glim/config
[2024-12-11 17:51:58.173] [glim] [info] load libodometry_estimation_gpu.so
[2024-12-11 17:51:58.353] [glim] [info] load libsub_mapping.so
[2024-12-11 17:51:58.354] [glim] [info] load libglobal_mapping.so
[2024-12-11 17:51:58.354] [glim] [info] load libmemory_monitor.so
[2024-12-11 17:51:58.355] [glim] [info] load libstandard_viewer.so
[2024-12-11 17:51:58.358] [glim] [info] load librviz_viewer.so
[ros2run]: Segmentation fault

**Environment (please complete the following information):**
 - CPU/GPU:  Core i9-13900HX/RTX4070
 - OS: Ubuntu 22.04
 - ROS: ROS2 humble
 - CUDA: CUDA 12.3
 - Build with CUDA: OFF

**Additional context**
Thanks

wsung1, 2024-12-11:
I’m experiencing the same issue.  
After completing a normal installation and using it without problems for a while, this issue suddenly occurred one day, and I still don’t know how to resolve it.

wsung1, 2024-12-11:
A bit different error I ran into was
terminate called after throwing an instance of 'thrust::system::system_error'
  what():  This program was not compiled for SM 75  
: cudaErrorInvalidDevice: invalid device ordinal

My environment is
CPU/GPU: Core i7-9700T/GTX 1650
OS: Ubuntu 22.04
ROS: Humble
CUDA: CUDA 12.3
Build with CUDA: ON

Efesendil, 2024-12-23:
Hello,
I have resolved this issue by installing a recent CUDA version, for me 12.6 works without any problem.

shavargo109, 2024-12-30:
> Hello, I have resolved this issue by installing a recent CUDA version, for me 12.6 works without any problem.

Installed 12.6 version but still run into the same issue, do you have any other potential solution?
Thanks
[image]

shavargo109, 2024-12-30:
I tried running gdb on glim_rosnode and found this issue
Thread 18 "glim_rosnode" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb9fff000 (LWP 42506)]
0x00007fffbab9d12b in std::vector<std::filesystem::__cxx11::path::_Cmpt, std::allocator<std::filesystem::__cxx11::path::_Cmpt> >::~vector() () from /usr/local/lib/libiridescence.so

it maybe something due to iridescence.so, but this file is generated by the installation guideline, I'm afraid it would be a bit difficult for me to solve this