koide3/glim #130: CMAKE Error: Project glim_ros tried to find library glim

Bexpie, 2024-12-28:
I get the following message when I try to `catkin build` the final step of the installation:

[image]

My environment is built as follows:
 - OS: WSL Ubuntu 20.04
 - ROS: Noetic
 - CUDA: 12.2
 - Build with CUDA: ON
 - CMAKE: 3.31.2

My suspicion are that one of the two things might be the cause:
1. When I ran `sudo ldconfig` I got the error: '/sbin/ldconfig.real: /usr/lib/wsl/lib/libcuda.so.1 is not a symbolic link'. To solve this, I renamed the existing libcuda.so.1 file.

2. I am unsure about the required package structure of my workspace. I currently have gtsam, iridescence and gtsam_points cloned at /home/user/ and the packages glim and glim_ros1 are located at /home/user/catkin_ws/src/

koide3, 2024-12-29:
Please try `catkin_make` instead https://github.com/koide3/glim/issues/114.

Bexpie, 2025-01-04:
Thank you for the reply, I finally managed to get it done! I tried `catkin_make` before, but some different error message made me switch to `catkin build`. I think my main issue was, that I didn't place the dependend packages outside the catkin workspace, as required. Maybe a little comment in the installation instructions will help users (with as little ROS experience as myself) in the future.

Also I want to mention some other issues I ran into during the installation process and how I solved them, with the intend of helping someone else:

- I manually installed CUDA 12.2 not the latest version 12.6
- I manually updated cmake to verison 3.31
- I manually installed Eigen 3.4 (not the latest version available on github) and placed a softlink for all include commands to work:
<pre><code>cd /usr/local/include/
sudo ln -s eigen3/Eigen Eigen</code></pre>

- For CUDA to be recognized, I added a new line with the path to my installed compiler in `/etc/environment`:
<pre><code>CUDACXX=/usr/local/cuda-12.2/bin/nvcc</code></pre>
- I set a environmental variable to pretend there being a higher version of OpenGL (e.g., v4.5 worked for me):
<pre><code>export MESA_GL_VERSION_OVERRIDE=4.5

# You might also want to put this line of code in:
sudo nano ~/.profile</code></pre>

closed 2025-01-04 by Bexpie