koide3/glim #138: /glim_ros/points containing no intensity field

boomer319, 2025-01-24:
**Problem**

ROS messages received from /glim_ros/points (which is publishing the preprocessed cloud?) do not contain the intensity field that is given in the raw input point cloud.

**Behavioral discrepancies**

The intensity can clearly be viewed in the glim viewer. Why is it not sent out via ROS under /glim_ros/points?
I need the lidar point cloud's intensity values to be sent out through the /glim_ros/points topic (for further processing based on intensity of the lidar measurments). Something to do with a CNN detecting shapes in the lidar's field of view.

**Environment**

 - CPU/GPU: AMD Ryzen 7 pro 5850u with radeon graphics
 - OS: Ubuntu 20.04
 - ROS: ROS1 noetic
 - CUDA: No
 - Build with CUDA: OFF
 - Point Cloud from Livox HAP TX solid state LiDAR

Did I miss something?

koide3, 2025-02-10:
I just pushed a patch to enable publishing point intensities https://github.com/koide3/glim/pull/144. Note that you need to update gtsam_points as well with https://github.com/koide3/gtsam_points/pull/48.

boomer319, 2025-02-10:
Thanks a lot! Really appreciate it!

closed 2025-02-10 by boomer319