koide3/glim #159: Mention CMAKE_CUDA_ARCHITECTURES in main documentation or compile for more architectures?

VRichardJP, 2025-02-20:
There is no mention of `CMAKE_CUDA_ARCHITECTURES` in the main documentation (https://koide3.github.io/glim/installation.html). 

If user cmake is recent enough (>= 3.24), then `CMAKE_CUDA_ARCHITECTURES=native` is used in `gtsam_points`, which definitely looks like a sensible default. Unfortunatelly, the current cmake on Ubuntu 22.04 is slightly older so `gtsam_points` falls back to compute capability >= 8.0 here: https://github.com/koide3/gtsam_points/blob/529d1d190067a0c3176275f4cbaeb6bd23575d9c/CMakeLists.txt#L72

Considering the popularity of old Jetson TX2/nano and cie (which support compute <= 5.3), maybe it could be worth to:

- mention the necessity to set `CMAKE_CUDA_ARCHITECTURES` for old GPUs in the main documentation
- compile for older compute version by default
- or compile for many arch by default (`BUILD_WITH_CUDA_MULTIARCH`)
- or stop compilation?

What do you think?

koide3, 2025-02-21:
OK, I think it would be good do add a few more options to the default `CMAKE_CUDA_ARCHITECTURES` (`53 62 87`) to butter support Jetson series.
https://github.com/koide3/gtsam_points/blob/529d1d190067a0c3176275f4cbaeb6bd23575d9c/CMakeLists.txt#L72

VRichardJP, 2025-02-21:
I think these are good defaults

koide3, 2025-02-21:
I just added several architectures to the default https://github.com/koide3/gtsam_points/pull/54. I'll soon merge it.

`set(CMAKE_CUDA_ARCHITECTURES 53 62 80 87 89 90 CACHE STRING "CUDA architectures" FORCE)`