koide3/glim #161: Georeferencing GLIM maps

VRichardJP, 2025-02-21:
I would like to georeference GLIM maps. There is already a `gnss_global` extension module which deals with GNSS data, but it does not fully satisfy my needs. In particular:

- The module uses Pose messages instead of NavSatFix, so GNSS sensor data must first be processed (e.g. with `navsat_transform_node`?). This is a bit inconvenient.
- The module computes the transformation `T_world_utm` only once, after the mapping device has travelled a few meter. So travelling longer distance won't improve the transformation (while new pose constraints are still added to the graph.)
- When the map size is not big (e.g. <1km) or when the path contains many loops, using poor GNSS data may be detrimental to the map. In other words, there are cases where I would rather not add any GNSS-based constraint to the graph (yet I would still like to georefence the map).

To address these issues, I am thinking about the following:

- Add support for NavSatFix messages. NavSatFix could be converted to UTM or MGRS with [geographiclib](https://geographiclib.sourceforge.io/2009-03/classGeographicLib_1_1UTMUPS.html). NavSatFix messages could be filtered based on fix status or position covariance.
- Use the first valid NavSatFix message to compute a GNSS origin and compute all `submap_coords` relatively to this origin point (thus avoiding errors on large floating point values).
- Add a `enable_gnss_factor`-like option. If enabled, use current logic to add pose factors to the graph (but use the GNSS origin instead of `T_world_utm`)
- Add a `on_update_submaps` callback, which receives submaps optimized by the main global mapping module and computes `T_world_utm` (using exactly the same logic than the current `T_world_utm` initialization). This `T_world_utm` transform is what is needed to georeference the map. There is no `save` callback yet, but I think it could be nice to dump the transformation data on save so that exporting a georeferenced map can be automated. Otherwise, the user could use tool like `pdal` to apply the transformation on the exported map manually.

koide3, 2025-02-27:
Yeah, the `gnss_global` module is just a simple PoC implementation and may not cover a wide range of applications. It's better to create another new module rather than extending the existing `gnss_global`.

I think it would be beneficial if the new module has the following features (mostly based on your proposal).
- Accepts both `NavSatFix` (global WGS84 coord) and `PoseStamped` (local reference coord).
- Continually optimizes `T_world_utm` along with sensor states.
- Has some mechanism to validate GNSS data correctness.

> There is no save callback yet, but I think it could be nice to dump the transformation data on save so that exporting a georeferenced map can be automated.

There is a [callback](https://github.com/koide3/glim/blob/ec802b895f3dad19e6253fcca03e2748e528bede/include/glim/util/extension_module.hpp#L28) for this purpose, and each extension module can save some information in the dump directory.

So, I think it is generally feasible to implement a module with the required functionalities. I wouldn't be able to work on this next moths because my bandwidth has been limited due to some assignments, but I can help you if you are willing to implement the new  GNSS module.

VRichardJP, 2025-02-27:
Thanks for the feedback.

I have started to implement a "`georef_global`" module, which filter `NavSatFix` messages, converts them to UTM frame, and adds interpolated pose/covariance as `gtsam::PoseTranslationPrior<gtsam::Pose3>` for each `X(submap->id)`. This pretty much works the same as in `gnss_global` except that `T_world_utm` is now a simple translation (I want to enforce world = utm + xyz_offset with the GNSS data). 

This kind of approach should work if this `georef_global` is the only module to add priors on the absolute position. However, `global_mapping` modules does add 2 constraints on the initial position:

- `gtsam::PriorFactor<gtsam::Pose3>(X(0), gtsam::Pose3::Identity(), gtsam::noiseModel::Isotropic::Precision(6, 1e6)`
- `gtsam_points::LinearDampingFactor(X(0), 6, params.init_pose_damping_scale)`

The first one sets the initial pose to 0. One simple trick I have used in the past is to use a high variance on initial x,y,z and yaw. Something like this:

```c++
Eigen::VectorXd init_noise(6);
//                 roll,     pitch,       yaw,       x,       y,         z
init_noise << 0.10*0.10, 0.10*0.10, M_PI*M_PI, 10.*10., 10.*10., 100.*100.;
model = gtsam::noiseModel::Diagonal::Variances(init_noise);
gtsam::PriorFactor<gtsam::Pose3>(X(0), gtsam::Pose3::Identity(), model)
```

However, I am not sure what `gtsam_points::LinearDampingFactor` is doing. What is its impact on the graph optimization?

> There is a [callback](https://github.com/koide3/glim/blob/ec802b895f3dad19e6253fcca03e2748e528bede/include/glim/util/extension_module.hpp#L28) for this purpose, and each extension module can save some information in the dump directory.

Ah yes, I think it should do the job.

koide3, 2025-03-06:
Thanks a lot for sharing your trial. I'll update the `gnss_global` module or create another one with the suggested ideas later.

VRichardJP, 2025-03-06:
There are a few caveats with the implementation above. Notably, `gtsam::Expression` does not appear to be serializable (or rather I don't know how to do it), so the `offline_viewer` will complain about not being able to restore the graph.

So I have replaced the inline `gtsam::Expression` by a new `gtsam_points::GeoRefFactor`, which does exactly the same (but is serializable). Implementation is rather straight-forward:
```c++

  virtual gtsam::Vector evaluateError(
    const gtsam::Pose3& T_geo_world,
    const gtsam::Pose3& pose_world,
    boost::optional<gtsam::Matrix&> H1 = boost::none,
    boost::optional<gtsam::Matrix&> H2 = boost::none) const override {

    // pose_geo = T_geo_world * pose_world
    gtsam::Matrix66 H_transform, H_pose;
    gtsam::Pose3 pose_geo = T_geo_world.compose(pose_world,
      H1 ? &H_transform : nullptr,
      H2 ? &H_pose : nullptr);

    // extract translation part
    gtsam::Matrix36 H_translation;
    gtsam::Point3 p_geo = pose_geo.translation(
      H1 || H2 ? &H_translation : nullptr);

    if (H1) {
      *H1 = H_translation * H_transform;
    }

    if (H2) {
      *H2 = H_translation * H_pose;
    }

    gtsam::Vector3 error = p_geo - p_geo_meas_;

    return error;
  }
```

There are a couple other modifications I had to make on GLIM side, such adding `T` key remapping, and adding a few `export_geo` functions to add the possibility to export georeferenced map from the `offline_viewer`.

One last problem I have yet to solve is to find what to do when multiple georeferenced maps are loaded: each has its own local `world` frame and own `T_geo_world`, but after the two maps are aligned I am not sure if it is still possible to use any of the `T_geo_world` to georeference the map ("moving" the local `world` frame basically invalidate the transform).

Nasekoma, 2025-04-14:
@VRichardJP, could you please publish your changes to your public forks of the mentioned repositories?

laurence-diack-pk, 2025-05-19:
> [@VRichardJP](https://github.com/VRichardJP), could you please publish your changes to your public forks of the mentioned repositories?

Here's one I've been working on based on the above comments:
https://github.com/laurence-diack-pk/glim_ext/tree/_LD_georef

Still haven't figured out the merging with multiple independently referenced maps though..

VRichardJP, 2025-05-19:
Merging multiple georeferenced maps can be done by transforming all point clouds into the same geo frame.

What GLIM currently does is:
- Load each map in its own local frame (`world_1`...`world_N`)
-  Merging is done by aligning `world_i` and `world_j`.

Instead, you can do the following:
- Use each session `T(0)` (renamed `T(0)`...`T(N)`) to transform the map points into the global geo frame (to increase precision, calculate a global offset on first geo referenced map and use shifted coordinates)
- Now all maps are already in the same geo frame, so the maps should be already almost aligned

gugelhop, 2025-06-05:
@LaurenceDiackPK
I tried your modification and it seems to be working. In the log, I could see that the georef constraints were added. However, I could not figure out how to export a georeferenced map to use with my vector map.
Could you please guide me through this step? 
Thanks.

Nasekoma, 2025-06-10:
@laurence-diack-pk 
I've been testing your extension and have found a few bugs in it:

1. Georeference uses keys based on values made with uppercase letters, but glim uses gtsam::symbol_shorthand functions, which use lowercase letters. So, in function Georeference::on_smoother_update all factors are always skipped because of the submap pose key

2. In the same function Georeference::on_smoother_update new factors can be based on new values that will be added in ISAM2 only after this function call. So, checking if all keys exist in the optimizer should also check if keys exist in the new_values

```cpp
        ...
        for (const auto& key : factor->keys()) {
            if (!isam2.valueExists(key) && !new_values.exists(key)) {
        ...
```

3. To export GeoPositionFactor serialize function needs to be written and exported. I've used pattern from the flat_earther extension and RotateVector3Factor from gtsam_points

```cpp
// in GeoPositionFactor
...
private:
  friend class boost::serialization::access;
  template<class ARCHIVE>
  void serialize(ARCHIVE & ar, const unsigned int /*version*/)
  {
    // NoiseModelFactor2 instead of NoiseModelFactorN for backward compatibility
    ar & boost::serialization::make_nvp("NoiseModelFactor2",
            boost::serialization::base_object<Base>(*this));
    ar & BOOST_SERIALIZATION_NVP(measurement_);
  }
...
// exporting in cpp file
BOOST_CLASS_EXPORT_GUID(glim::GeoPositionFactor, "glim::GeoPositionFactor");
```

4. I suspect using ECEF makes T_geo_world too unstable because of big values -- gtsam::IndeterminantLinearSystemException is thrown almost immediately. I changed ECEF to local Cartesian and added LinearDampingFactor for T_geo_world. This helped in my case, but the resulting map is much worse than without GNSS data (probably needs better GNSS data filtration and other parameters).