koide3/glim #165: Minor memory leak in `OdometryEstimation*`

noelex, 2025-02-27:
While reading the code of `OdometryEstimationIMU` class, I noticed a minor memotry leak. It seems that the `frames` vector will only ever grow but never shrink. That is, an `EstimationFrame` is freed during marginalization but its slot in `frames` is not:
https://github.com/koide3/glim/blob/ec802b895f3dad19e6253fcca03e2748e528bede/src/glim/odometry/odometry_estimation_imu.cpp#L330-L332

If I understand correctly, this will cause slow memory leak depending on the frame rate of the LiDAR (e.g. 160 bytes per second for a 10 Hz LiDAR), which might be prolematic if the application needs to run for a long duration (e.g. localization).

koide3, 2025-03-06:
I've noticed this memory leak, and I consider it is tolerable as it takes only up to a few hundreds MB for 24 hours. While it should be easy to fix this, it has a low-priority for now. If you're facing problems in real use scenarios (e.g., running the system over a few days), we can consider fixing it sooner.

noelex, 2025-04-01:
Just pointing out in case someone tries to use `OdometryEstimationImu` for localization. Closing this for now as this shouldn't be a problem for mapping.

closed 2025-04-01 by noelex