koide3/glim #168: Unstoppable bumping errors

GuoZheXinDeGuang, 2025-03-02:
When I ran a rosbag to convert it into the format of glim, it went to unstoppable errors when I do the conversion. Is this an issue related to the parameters? Thank you. 

Here is the picture:

[image]

koide3, 2025-03-06:
It indicates that the global mapping graph is in an unstable condition due to some reasons, e.g., less overlap between submaps. Please try setting `config_global_mapping_gpu.json` as follows to make the global mapping create relative pose factors between submaps that would stabilize the optimization process.

```
    "create_between_factors": false,
    "between_registration_type": "GICP",
```

GuoZheXinDeGuang, 2025-03-11:
The problem still exists. From the offline viewer, there is great Z-drift at around 20 percent, and after that the errors start bumping.