koide3/glim #176: Coordinate system in GLCanvas is flipped?

VRichardJP, 2025-03-14:
I am trying to display a point cloud using `guik::GLCanvas`. 

Inspired by the code from GLIM, I have come up with something like this:

```c++
  if (!flat_map_canvas_) {
    flat_map_canvas_.reset(new guik::GLCanvas(Eigen::Vector2i(width, height)));
    flat_map_canvas_->camera_control.reset(new guik::TopDownCameraControl());
  }

  ImGui::BeginChild(
    "flat_map",
    ImVec2(width, height),
    false,
    ImGuiWindowFlags_ChildWindow | ImGuiWindowFlags_AlwaysAutoResize | ImGuiWindowFlags_NoMove | ImGuiWindowFlags_NoTitleBar | ImGuiWindowFlags_NoSavedSettings |
      ImGuiWindowFlags_NoNavFocus);

  if (ImGui::IsWindowFocused()) {
    flat_map_canvas_->mouse_control();
  }

  flat_map_canvas_->bind();

  flat_map_canvas_->shader->set_uniform("model_matrix", Eigen::Matrix4f::Identity().eval());
  flat_map_canvas_->shader->set_uniform("color_mode", guik::ColorMode::VERTEX_COLOR);
  
  flat_map_buffer_->draw(*flat_map_canvas_->shader);

  flat_map_canvas_->unbind();

  ImGui::Image((void*)flat_map_canvas_->frame_buffer->color().id(), ImVec2(width, height));

  ImGui::EndChild();
```

Where the "flat_map" is simply flatten submaps:

```
  std::vector<Eigen::Vector4d> flat_points;
  flat_points.reserve(nb_points);
  for (const auto& submap: submaps) {
    for (int i = 0; i < submap->frame->num_points; ++i) {
      flat_points.emplace_back(submap->T_world_origin * submap->frame->points[i]);
    }
  }

  std::vector<Eigen::Vector4d> flat_colors(nb_points, Eigen::Vector4d{0,0,0,0}); // default color = RED

  auto flat_map = std::make_shared<gtsam_points::PointCloudCPU>();
  flat_map->add_points(flat_points);
  flat_map->add_colors(flat_colors);
```

The map points looks ok, except the view is flipped:

[image]

I have tried both the top to down and xy orbit cameras, in both cases everything look like as if the camera was looking from below the map.

I have checked at other GLIM modules displaying pointclouds in a `GLCanvas`, and in GLIM node too the submap pointcloud looks flipped:

[image]

Here is the default view matrix when using `guik::TopDownCameraControl`:

```
  0  -1   0  -0
  1   0   0  -0
 -0  -0   1 -80
  0   0   0   1
```

[image]

And here the default projection matrix of the canvas `guik::BasicProjectionControl`:

```
3.73205       0       0       0
      0 3.73205       0       0
      0       0  -1.002  -2.002
      0       0      -1       0
```

Both look normal to me.

koide3, 2025-03-21:
IIRC, this stems from the difference of image coordinates between ImGui (top left = origin) and OpenGL (bottom left = origin), and a workaround is to specify the texture coordinates when calling `ImGui::Image` as in the following line.

https://github.com/koide3/iridescence/blob/0aa860d5fdfd568c12d21a08c19321620621a8a2/src/guik/viewer/light_viewer_context.cpp#L150

VRichardJP, 2025-03-24:
`ImVec2(0, 1), ImVec2(1, 0)` does the trick! Thank you!

closed 2025-03-24 by VRichardJP