koide3/glim #183: glim: GLFW 65539 error and segmentation fault on Jetson Orin with ROS 2 Docker

kimboseong0602, 2025-03-24:
### 🧩 Environment

- **Platform**: Jetson Orin (aarch64)
- **ROS 2**: Humble
- **Docker**: Yes (running on Jetson)
- I have a custom-built Docker image with:
  - ROS 2 + required dependencies
  - all `glim` dependencies preinstalled
  - built natively for `aarch64` (Jetson Orin)

---

### ❗ Problem Summary

I'm trying to run `glim` in a multi-container ROS 2 setup. Here's the situation:

- **Container A**: runs a Livox Mid-70 LiDAR node, visualized in RViz2
- **Container B**: runs `glim` SLAM
- ROS 2 Domain ID and DDS communication between containers is working (confirmed)
- Topic `/livox/lidar` is visible and correctly configured in `config_ros.json`
- `lidar_frame_id` is also set to match `livox_frame`

---

### ⚠️ The Issue

When I run `glim_rosnode` inside Container B, I get the following error:

GLFW error 65539: Invalid window attribute 0x0002000D

diff
Copy
Edit

So I added **X11-related Docker options**:

```bash
-e DISPLAY=$DISPLAY
-v /tmp/.X11-unix:/tmp/.X11-unix
--runtime=nvidia
--privileged
--net=host
Inside the container:

nvidia-smi and nvcc -V both work

glfw still throws error 65539

No display output from glim (just a black window)

After I close the window or kill the process manually, I see this log:

css
Copy
Edit
[odom] [info] estimate initial IMU state
Segmentation fault
Also, right before it exits, I briefly (less than a second) see the livox_frame appearing in the window — and then it closes.

✅ What I’ve tried
X11 settings and xhost +local:root on the host

Confirmed topics via ros2 topic list and types

Checked config_ros.json: topic set to /livox/lidar

Ensured lidar_frame_id is correct

GLFW version is 3.3.6 (which does not support GLFW_MOUSE_PASSTHROUGH)

Tried patching glfwWindowHint() calls conditionally (still same result)

🙏 Questions
Is there anything in glim_rosnode that requires GLFW 3.4 or newer?

What are the required conditions for the viewer window to stay open?

Could the segmentation fault be caused by a missing or invalid IMU input?

Any known issues when running glim headless (in Docker)?

Thanks in advance for any guidance!

kimboseong0602, 2025-03-24:
If you need any additional information, I'd be happy to provide the files.🙂

kimboseong0602, 2025-03-24:
here is recording video!!

https://github.com/user-attachments/assets/0e95c5e4-e083-4db5-92fc-98b7ab7ce7d8

kimboseong0602, 2025-03-24:
After connecting the IMU sensor, the topic is being published successfully. Thx!!

Yccmv, 2025-08-19:
Hi @kimboseong0602 

I'm glad to see you solved the segmentation fault by connecting an IMU sensor.

I am a beginner trying to do the same thing with the Livox Mid-70, but I cannot find any documentation on how to configure GLIM with an IMU. Could you please share some details about your setup?

1. **Hardware**: Which IMU model are you using, and how is it physically connected  ?
2. **Software**: Which ROS driver are you using to publish the IMU data?
3. **GLIM Config**: What is the ROS topic name and message type that GLIM expects for the IMU?
4. **GLIM Files**: Did you need to change any configuration files in GLIM to enable it?

Any help you could provide would be greatly appreciated.

Thank you!