koide3/glim #197: bad localization when using ouster os1 lidars and livox avia

aktamaa, 2025-04-15:
Hello @koide3 and team.

First of all, thanks a lot for sharing this wonderful work ^^.
I am new to robotics and wanted to know how to correctly use your package with ouster os1 lidars (32 and 128 channels) and livox avia. I tried to use it with livox mid360 and everything is working really fine and straight forward. But when I try to use it with ouster os1 lidars and livox avia, none of them are working as expected. The localization result was not good as shown in the video I attached below. What I notice when I am run it is there some warning `large time gap between consecutive LiDAR frames!!` and `large time difference between points and imu!!`, then the error `an indeterminant lienar system exception was caught during global map optimization!!`, and warning `insert a damping factor at x1 to prevent corruption` comes after some rotation done physically to the lidar.

This is how I run the glim node, which actually using os0 config. For the livox avia, I am using the same config I use for the livox mid360.

`ros2 run glim_ros glim_rosnode --ros-args -p config_path:=$(realpath ~/ros2_ws/src/glim/config_versatile/os0)`
`ros2 run glim_ros glim_rosnode --ros-args -p config_path:=$(realpath ~/ros2_ws/src/glim/config_versatile/livox)`

Thank you in advance for any help or suggestion!

https://github.com/user-attachments/assets/debb21a3-8b17-4420-b7ff-4941d754488f

aktamaa, 2025-04-17:
Updating this, I found that the reason of the drift happens when using the ouster lidar was simply because the coordinate frame of the imu and lidar in the sensor was changed in the ouster ros driver. The lidar frame was rotated by 180 degree in the z-axis so by modifying the T_lidar_imu like this below, everything went smoothly. I was following the transformation provided in the config_sensors file in the provided config_versatile which is actually not considering the rotation of the lidar frame, so I think it might be good to update it so that the package will be easier to be used by a newbie like me XD, and more straight forward to be implemented. For the LIVOX, I haven't tried it again

Suggested T_lidar_imu in the config_sensor.json file:
```
"T_lidar_imu": [
      0.006,
      -0.012,
      0.008,
      0.0,
      0.0,
      0.0,
      1.0
    ]
```

The actual T_lidar_imu that should be used:
```
"T_lidar_imu": [
      0.006,
      -0.012,
      0.031,
      0.0,
      0.0,
      1.0,
      0.0
    ]
```

koide3, 2025-04-18:
Looks good! Ouster changed the LiDAR-IMU transformation at some point, and thus newer Ouster LiDARs need a different setting.

Note that errors `large time gap between consecutive LiDAR frames!!` indicate LiDAR data are dropped due to communication issues (it occasionally happens with ROS2). I recommend try composable nodes to reduce the communication load.

closed 2025-06-04 by aktamaa

aktamaa, 2025-06-04:
@koide3 . Understood. Do you have any suggestion on how to overcome this problem and is it something related to the DDS? I am currently working with the MID-360 actually and there is no such problem with it, but maybe in the future I will need to work with Ouster again. Please kindly provide some within your free time. Thank you so much! ^^