koide3/glim #215: An error occurred while merging the maps

casso1993, 2025-05-15:
Hello，great job on this project!

An error occurred when I merged two maps. I have repaired broken map first, but still have problems when I find overlapping submaps. Thanks for the help.

[image]

[image]

casso1993, 2025-05-20:
The error may be caused by excessive GPU memory usage. I used 3060 6G laptop to build a map, but the map was too large. When the GPU memory usage reached 5.9G, an error occurred. How can I fix it?

koide3, 2025-05-28:
The GPU RAM usage can be mitigated by changing the following params:
- Set  `submap_target_num_points` in `config_sub_mapping_gpu.json` to 10000 ~ 25000 (# points for each submap)
- Set `min_implicit_loop_overlap` in `config_global_mapping_gpu.json` to 0.3 ~ 0.4 (Reducing # factors in global optimization)

I'm also thinking of implementing some additional mechanisms to reduce the memory usage.