koide3/glim #216: Error "double free or corruption" when starting glim_rosnode on Jetson Orin AGX in docker

JeremyLebon, 2025-05-21:
**Describe the bug**
The error "double free or corruption" when starting glim_rosnode is received.

[image]

**Expected behavior**
Normally the glim_rosnode should be starting. It already works on my laptop with a nvidia gpu.

**Screenshots**

I added the debug parameter and get the following

[image]

I can see with jtop everthing is available inside the docker container.

[image]

I already tried version 1.0.6 and the latest version. I already changed the config from gpu to cpu. But still the same error. 

**Environment (please complete the following information):**
 - CPU/GPU: Jetson Orin AGX
 - OS: 22.04 
 - ROS: ROS 2 Humble
 - CUDA: 12.2
 - Build with CUDA: ON
 - Docker is used
 - Jetpack 6.1

Any suggestion where I should look?  Is there a example of docker file for a Jetson with GLIM and ROS2?

koide3, 2025-05-28:
I suppose the issue is caused by mismatch of CUDA arch. Can you run `cuobjdump /usr/local/lib/libgtsam_points_cuda.so` to see what architectures are supported. If something like `sm_87` is missing, please try adding "-DCMAKE_CUDA_ARCHITECTURES=87" when building gtsam_points.

JeremyLebon, 2025-06-04:
Sorry, i made a mistake with the Jetpack version. I was using the r36.3 available at the [Jetson Interference](https://github.com/dusty-nv/jetson-inference) repo. But this version is outdated and doesn't match with glim.

I now have build a image with the help of the script at the [jetson container](https://github.com/dusty-nv/jetson-containers) repo. And it know works like charm. :) (after struggling with realsense camera)

[image]

closed 2025-06-04 by JeremyLebon