koide3/glim #217: glim_rosbag occurs segmentation fault

atinfinity, 2025-05-22:
**Describe the bug**

I run `glim_rosbag` with CUDA.
But,  it sometimes causes segmentation fault.

```bash
[ros2run]: Segmentation fault
```

**To Reproduce**

```bash
ros2 run glim_ros glim_rosbag <ROSBAG_PATH> --ros-args -p config_path:=/glim/config/livox/
```

I download `config_versatile.tar.gz` from <https://koide3.github.io/glim/demo.html>.

**Expected behavior**

`glim_rosbag` finished without segmentation fault.

**Environment (please complete the following information):**
 - CPU/GPU: Intel Core i7-13700HX
 - OS: Ubuntu 24.04
 - ROS: ROS 2 Jazzy
 - CUDA: CUDA 12.6
 - Build with CUDA: ON

**Additional context**

I run  `glim_rosbag` with gdb.

```bash
Thread 19 "glim_rosbag" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc27de000 (LWP 5931)]
operator() (factors=..., target_key=<optimized out>, source_key=<optimized out>, source_key@entry=8646911284551352639,
    target=std::shared_ptr<const glim::EstimationFrame> (empty) = {...}, source=std::shared_ptr<const glim::EstimationFrame> (use count 3, weak count 0) = {...},
    __closure=<optimized out>) at /home/jazzy/ros2_ws/src/glim/src/glim/odometry/odometry_estimation_gpu.cpp:144
144         for (const auto& voxelmap : target->voxelmaps) {
```

I checked `target` is nullptr when `glim_rosbag` causes segmentation fault.

koide3, 2025-05-28:
Hmm, that's weird. `target` will never be `nullptr` in a usual way. Could you add `-p debug:=true` and share `/tmp/glim_log.log` and `glim_odom.log` ?

atinfinity, 2025-06-02:
Thank you for your comment. I attached log.
[glim_log.log](https://github.com/user-attachments/files/20545921/glim_log.log)
[glim_odom.log](https://github.com/user-attachments/files/20545987/glim_odom.log)

atinfinity, 2025-06-10:
Thank you for your advice. I resolved this problem.

> I recommend fixing the timestamp issue, e.g., by re-configuring communication settings (e.g., using composition) or re-ordering messages in the rosbag based on data timestamps (i.e., msg.header.stamp).

I tuned Fast DDS for large data. As a result, it seems that issue of rosbag recording is resolved.

closed 2025-06-10 by atinfinity