koide3/glim #220: Debain Package Release Detials

qiayuanl, 2025-06-04:
Hi, both your code quality and software engineering are excellent! Thank you for the open source of this project.

I am curious about how you packaged the Debian package with different CUDA support. 

I am currently using [ros-deb-builder-action](https://github.com/jspricke/ros-deb-builder-action) in my GitHub build farm: [legged_buildfarm](https://github.com/qiayuanl/legged_buildfarm). 
It basically does `bloom-generate` and `fakeroot debian/rules`.

Do you have any advice for releasing a package with different CUDA versions and without CUDA?

koide3, 2025-06-09:
Because I only provide self-built binaries on our own PPA, I'm not sure about a good way to build Debian packages. In my case, I manually re-write configurations to support different CUDA versions (it's certainly a bad way). Please let me know if you figure out how to do it in a better way.

qiayuanl, 2025-06-11:
Can you share some script for your own PPA? It would be very helpful for me to know how you did it.

koide3, 2025-06-15:
Here you can find the scripts for generating deb files: https://github.com/koide3/ppa/tree/main/packages