koide3/glim #236: Lidar Points moving opposite direction to the robot direction

vickzk, 2025-07-01:
**Robot Type:** Mobile robot
**Sensor Setup:** Separate LiDAR and IMU sensors
**Transform (T_lidar_imu):** Set as identity (translation: zero, rotation: [0, 0, 0, 1])
Data Flow: All sensor data is transformed to the `base_link` frame and passed to GLIM

**Issue Observed:**
While running SLAM with GLIM, the map appears accurate, and the odometry is mostly stable. However, there is a noticeable drift behavior during continuous motion:

As the robot moves forward, the LiDAR point cloud appears to move backward relative to the robot’s motion.

This drift accumulates and remains while the robot is in motion.

Once the robot stops, the LiDAR data snaps back or settles into the correct pose.

**Request for Help:**
What could be causing this drift during motion and sudden correction on stopping? Any insights or suggestions to resolve this behavior would be appreciated

koide3, 2025-07-07:
There may be IMU-related issues. Can you check if 1) the LiDAR-IMU is correct, 2) IMU data has a reasonable frequency (about 100Hz)?

vickzk, 2025-07-07:
I've verified the TF and it looks correct. Regarding the IMU rate, I'm running in simulation with a configured update rate of 100 Hz. However, due to Gazebo's real-time factor, the effective rate I'm observing is around 90 Hz.